arduino/arduino-ide #1554: [Arduino 2.0.0]: Language support

hasenradball, 2022-10-11:
### Describe the problem

I use the Arduino IDE 2.0 and switche to German.

But I saw only some Menu items was switched to german...
Half of the Menu ites in english, halb of menue item in german...
It makes not a good picture.
It seems that the software has some problems.

### To reproduce

Just try it

### Expected behavior

All menue items are switched to the corresponding language.

### Arduino IDE version

2.0.0

### Operating system

macOS

### Operating system version

latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-11:
Hi @hasenradball. Thanks for your report. Please try the nightly build of the Arduino IDE (as you were instructed to do already but evidently ignored :unamused:). There was a problem like this in Arduino IDE 2.0.0, but that problem was fixed two weeks ago: https://github.com/arduino/arduino-ide/pull/1449. 

The download links for the nightly build are listed here:

https://github.com/arduino/arduino-ide#nightly-builds

hasenradball, 2022-10-11:
Hi yes ok I see, sorry.

But I have an other Problem with the IDE 2.0...
I am not able to install it easily on my ubuntu Computer like the version 1.0

With the App Image it does not work.
And using the ZIP I have no icon to start the IDE.

Your Online Docu does not help here....

per1234, 2022-10-12:
Hi @hasenradball. Thanks for the update.

I'm sorry you are having trouble installing the Arduino IDE. Please post a detailed description of the problems you are having over on the Arduino Forum. We'll be sure to help you out over there:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

closed 2022-10-12 by per1234