arduino/arduino-ide #1556: After changing Preferences:Sketchbook location, some libraries are not found

lilbuzz, 2022-10-12:
### Describe the problem

When I add "\Sketches" to the default Sketchbook location in the preferences, I see the following error when I attempt to compile the project:
C:\Users\<username>\Documents\Arduino\Sketches\<sketchname>\<sketchname>.ino:35:34: fatal error: UniversalTelegramBot.h: No such file or directory
compilation terminated.

exit status 1

Compilation error: UniversalTelegramBot.h: No such file or directory

### To reproduce

(include UniversalTelegramBot library)
Open File-Preferences
Add \Sketches to the Sketchbook location, and click OK.
Compile project

### Expected behavior

Library should be found, and compile should work.
If I remove "\Sketches" from the Sketchbook location, my projrct compiles correclty, even though the project is in the Documents\Arduino\Sketches folder.

### Arduino IDE version

2.0.0

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-12:
Hi @lilbuzz. Thanks for your report.

This is the expected behavior. The Arduino IDE installs libraries to the `libraries` subfolder of the path set in the "**Sketchbook location**" preference. This means that when you change the "**Sketchbook location**" preference, all the libraries you installed with the previous setting are no longer available to the Arduino IDE. You will need to either reinstall the libraries or else move the existing `libraries` folder from the old sketchbook location to the new one.

lilbuzz, 2022-10-12:
Why is this location labeled Sketchbook location, if it used for a different purpose?
Perhaps there needs to be two locations.