arduino/arduino-ide #1560: Strange problem

bozzerjoseph, 2022-10-14:
### Describe the problem

If in this line of writing in the dac I write "int" in front of "valeuraenvoyer" .. the sketch is compiled and I televerse it without problem

`// Calculation of the value corresponding to the desired voltage at the output of the DAC and its shipment
  int Valeuraenvoyer = tensionvoulueensortie * 1024/3.3;

`
On the other hand if I write "Float" in front "Valeuraenvoyer" .. The sketch is compiled but impossible to tending it
I have this error without any explanation:

`Write 44392 bytes to flash (694 pages)

[==] 9% (64/694 pages)
[=====] 18% (128/694 pages)
[======= "27% (192/694 pages)
[==========/36% (256/694 pages)
[============== "(320/694 pages)
[================= "(384/694 pages)
[===================== "(448/694 pages)
[=======================/73% (512/694 pages)
[=========================/82% (576/694 pages)
[============================= "(640/694 pages)
[=============================/100% (694/694 pages)
Done in 0.256 seconds

VERIFY 44392 bytes of flash with checksum.
VERIFY FAILED
Failed Uploading: Uploading Error: Exit Status 2
`
Same if instead of putting Float in this line, I place at the top of the sketch:

Float Valeuraenvoyer = 0;

### To reproduce

If I place "int" i can upload into my arduino..... Immediately I replace it with "float" and I have a mistake! ... Imediatement I put back "int" and everything is fine ... I put it back and I have mistakes again

### Expected behavior

I expect to upload my sketch

### Arduino IDE version

  IDE 2.0.0  date 2022-09-14 

### Operating system

macOS

### Operating system version

Mojave

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-14:
Hi . Thanks for your interest in this open source project. This issue tracker is only to be used to report bugs or feature requests specific to the project. This topic is more appropriate for the Arduino Forum. I'm sure we will be able to help you out over there:

https://forum.arduino.cc/

bozzerjoseph, 2022-10-14:
Hi

I already posted my problem there and they think it must be a bug in IDE and sent me here because for them replacing int with float is completely normal and allowed... it shouldn't prevent the upload

Regards

https://forum.arduino.cc/

per1234, 2022-10-14:
Thanks. I found it and will proceed with the investigation over there. If we find it is caused by a defect in the Arduino IDE codebase I will be happy to reopen this.

per1234, 2022-10-15:
Unfortunately we did not make any progress in identifying the cause on the forum.

I compared the upload command shown in the output from each IDE version used by @bozzerjoseph and they are identical:

Arduino IDE 1.8.19:

```text
Le croquis utilise 43724 octets (16%) de l'espace de stockage de programmes. Le maximum est de 262144 octets.
Redémarrage forcé en cours en 1200bps ouvert/fermé sur le port /dev/cu.usbmodem14344201
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } => {}
PORTS {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmodem14344201, /dev/tty.APPLE-WirelessiAP, /dev/tty.Bluetooth-Incoming-Port, /dev/tty.usbmodem14344201, } / {/dev/cu.APPLE-WirelessiAP, /dev/cu.Bluetooth-Incoming-Port, /dev/cu.usbmod [... cut ...]
As the problem does not occur when @bozzerjoseph uses Arduino IDE 1.8.19 and does occur when using 2.x, it seems that the IDE is somehow a factor in this checksum failure. Unfortunately I don't have any idea of what that would be and investigating it will be difficult since I can't reproduce the problem.

I suspected that the Serial Monitor might somehow be interfering with communications during the upload or verification (along the lines of https://github.com/arduino/arduino-ide/issues/375). However, [@bozzerjoseph says that the Serial Monitor is not open when the problem occurs](https://forum.arduino.cc/t/stockage-non-volatil/1038996/127).

closed 2022-10-17 by per1234