arduino/arduino-ide #1561: Keybinding to browse different frames/views

josx, 2022-10-14:
### Describe the problem

I am working with a blind programmer that usually use IDEs. So We need some Accesibility by keys to browse all the frames/views on the IDE. 

The bigger problem is on **OUTPUT** view becasuse you can get focus there by keys. (so cant listen errors on boards, compile or uploading code).

### To reproduce

Touch any keys to try got focus on OUTPUT view.

### Expected behavior

A new keybinding to browse all graphics widget (icons, view, etc)

### Arduino IDE version

e577de4e8e8a3682ea4422fdb38658eb30628375

### Operating system

Linux

### Operating system version

Ubutnu focal

### Additional context

I think that would be great to have one key to browser like a carousel all the graphic widgets (so blind readers can read whatever are on it).

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-10:
Thanks for your report @josx

>we need some accessibility by keyboard to browse all the frames/views on the IDE.

I did some experimentation using [**Windows Narrator**](https://support.microsoft.com/en-us/windows/complete-guide-to-narrator-e4397a0d-ef4f-b386-d8ae-c172f109bdb1) and found I was able to access all panels and views using **Narrator + Right arrow** / **Narrator + Left arrow** (Up/Down arrow when in "Scan Mode").

I found the same with [**NVDA**](https://www.nvaccess.org/) in "Browse Mode".

Unfortunately I did not have any luck with navigating the Arduino IDE UI using [**Orca**](https://help.gnome.org/users/orca/stable/index.html.en), but it is possible I simply don't know how to use it.

Is there a specific screen reader you are using?

> The bigger problem is on OUTPUT view because you can't get focus there by the keyboard

I found a somewhat roundabout way to accomplish this. I don't think it counts as a real solution, but perhaps it will be useful as a workaround so I'll share it anyway:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the the "**Toggle Output View**" command from the menu.
   You can quickly find it by typing the first characters of `toggle output view`.
   After executing the command the first time, the command will be listed at the top of the **Command Palette** menu, so you will be able to quickly select it without needing to type the command.
1. Press the <kbd>**Enter**</kbd> key to execute the command.
1. If the "**Output**" view was not previously active, it will now have focus. If it was previously active, the bottom panel will have collapsed. If the latter, repeat the steps above once more to reopen the bottom panel and get focus on the "**Output**" view.

The same technique can be used for the "**Serial Monitor**" view, though the **Tools > Serial Monitor** menu item or its keyboard shortcut might be more straightforward to the users.

If you want to avoid needing to use **Command Palette**, you can assign a keyboard shortcut to the "**Toggle Output View**" command (it actually has one by default but it collides with the "**Upload Using Programmer**" shortcut (https://github.com/arduino/arduino-ide/issues/1658). I'll provide instructions for doing that:

1. Select **File > Advanced > Keyboard Shortcuts** from the Arduino IDE menus.
   A "**Keyboard Shortcuts**" view will open in the main panel.
1. In the "**Search keybindings**" field of the view, type `Toggle Output View`
1. Select the "**Edit Keybinding**" icon to the left of the "**Toggle Output View**" command row in the table (it seems the icon will be invisible when selected via a keyboard due to requiring a mouse hover to appear).
   An "**Edit Keybinding for output:toggle**" dialog will now open.
1. Type the keyboard shortcut you want into the field in the dialog.
1. Press <kbd>**Tab**</kbd> to select the **OK** button.
1. Press the <kbd>**Enter**</kbd> key.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**W**</kbd> keyboard shortcut to close the "**Keyboard Shortcuts**" view.

per1234, 2022-11-10:
I see now there is actually a quite usable way to access the views in the bottom panel (including "**Output**") so long as the bottom panel is expanded (which it would typically be when the user wanted to access it since the IDE automatically expands it as appropriate):

1. If the IDE is not already in the "tab key moves focus" mode, execute the "**Toggle Tab Key Moves Focus**" command (either via Command Palette or keyboard shortcut).
1. Press tab until the bottom panel has focus.
1. If another view is selected in the bottom panel, use <kbd>**Ctrl**</kbd>+<kbd>**Tab**</kbd> to select the "**Output**" view.

In case the bottom panel is collapsed, the "**Toggle Output View**" command should be used instead, per my instructions above. Or for those who prefer using the primary UI, there is a "**Toggle Bottom Panel**" icon on the right side of the status bar. That icon is not accessible via the IDE's native navigation, but can be accessed using the enhanced navigation provided by **Narrator** and **NVDA**.

The reason I considered my instructions above to be more of a workaround than a solution is because they required running the command twice to untoggle and then retoggle the view in order to get the side effect of obtaining focus on the view, which is clearly a hack. I don't see any problem with using the command as intended simply to toggle the view when that is needed.