arduino/arduino-ide #1572: Use more efficient method to get Arduino CLI version

per1234, 2022-10-21:
Use more efficient method to get Arduino CLI version
### Describe the request

Employ a technique for obtaining the version of the bundled Arduino CLI that does not trigger the pointless update check.

🙂 The **Help > About Arduino IDE** menu item will be responsive.

### Describe the current behavior

The **Help > About Arduino IDE** menu item opens a dialog which displays information about the Arduino IDE. This includes the version of the bundled copy of Arduino CLI.

The Arduino CLI version is obtained by running the following command:

```text
arduino-cli version --format json
```

In addition to displaying the version, this command triggers a check for the availability of a newer version of Arduino CLI:

https://github.com/arduino/arduino-cli/blob/5efa9b5d3596d4dd88f16813d1018f3f60481b05/cli/version/version.go#L61

While that update check is useful to the users running the command directly, it is completely pointless for this application, both because the IDE does not even use the information about availability of an update and because the IDE is designed to use a specific version of Arduino CLI and updating the bundled Arduino CLI to a new version is strongly discouraged.

For unknown reasons (perhaps transient network conditions), this update check sometimes takes on the order of a few seconds to complete.

🙁 The dialog opening is blocked while waiting, which is a poor experience for the user who will frustrated that such a simple dialog should be so slow to open.

### Arduino IDE version

2.0.1-snapshot-32d904c

### Operating system

Windows

### Operating system version

10

### Additional context

The update check will not be done when a non-release version of Arduino CLI is bundled with the IDE, so make sure your IDE meets that condition if trying to reproduce the slow dialog opening.

---

The slow update check is intermittent. It does happen frequently though so you should be able to reproduce it within several cycles of opening and closing the dialog.

---

Previously discussed in passing at https://github.com/arduino/arduino-ide/issues/1289#issuecomment-1205469774

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2022-10-24:
IDE2 wants to use the command line interface to get the version. It must work when something bad happens with the core gRPC client, or there is no connection to the daemon. I wrote the reasons here: https://github.com/arduino/arduino-ide/issues/1289#issuecomment-1206131063. The version must be available. Otherwise, users cannot specify the details when submitting a bug report.

I agree that the remote version check is not needed. CLI should provide a flag not to perform the update check when simply printing the version, and IDE2 will start using this new flag.

per1234, 2022-10-24:
An alternative would be to just get it from the `arduino.cli.version` field of `arduino-ide-extension/package.json`.

That won't be correct if someone swapped out the `arduino-cli` executable from the IDE installation, but the likely of that actually being a real world problem is incredibly low:

So far, there has never been a single time I found the need for this Arduino CLI version information in years of supporting IDE users. From a purely statistical point of view, this indicates that the information will be needed only rarely.

I have not seen any evidence that users are swapping out the Arduino CLI executable, nor any evidence that they are interested in doing this. So the chances of the Arduino CLI version not matching that in the `package.json` is very low.

The users who would swap out the Arduino CLI executable will be more advanced users, who are more likely to understand this is significant and mention it in support requests and bug reports.

kittaakos, 2022-10-25:
I am OK with removing the commit hash of the CLI if @ubidefeo is OK with it. Originally, the CLI details were requested to be part of the `About` dialog in ATL-663.

per1234, 2022-11-26:
Since we didn't receive any feedback after a month, I'll give the approval, for the reasons I explained above.

closed 2023-01-11 by kittaakos