arduino/arduino-ide #1573: IDE cannot start on Ubuntu Ubuntu 22.04.1 LTS

GTorreil, 2022-10-23:
### Describe the problem

IDE crashes on launch on Ubuntu 22.04.1 LTS

### To reproduce

1. Extract IDE 2.0.0 zip
2. Launch it :
``` shell
➜  arduino_2.0.0 ./arduino-ide 
[1023/142406.549593:FATAL:v8_initializer.cc(447)] Error loading V8 startup snapshot file
[1]    179508 trace trap (core dumped)  ./arduino-ide
```

### Expected behavior

IDE Launches

### Arduino IDE version

2.0.0

### Operating system

Linux

### Operating system version

Ubuntu 22.04.1 LTS

### Additional context

The AppImage works fine, but it makes it challenging to access the Arduino executable from external tools. 

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-10-24:
What's the output of `uname -a` on your machine? Which download link did you use?

GTorreil, 2022-10-31:
I'm sorry , I missed the notification for your reply. 
I'll give feedback shortly.

GTorreil, 2022-11-01:
``` shell
➜  ~ uname -a
Linux guillaume-MS-7C56 5.15.0-52-generic #58-Ubuntu SMP Thu Oct 13 08:03:55 UTC 2022 x86_64 x86_64 x86_64 GNU/Linux
```

Download link : https://www.arduino.cc/en/software > linux zip file

GTorreil, 2022-11-18:
Good evening, good news :)

I downloaded yesterday's 2.0.2 release ([arduino-ide_2.0.2_Linux_64bit.zip](https://github.com/arduino/arduino-ide/releases/download/2.0.2/arduino-ide_2.0.2_Linux_64bit.zip)).
This one just works. 

Issue solved I guess :)

closed 2022-11-18 by GTorreil

kittaakos, 2022-11-18:
Thank you so much for informing us and taking care of this open issue.