arduino/arduino-ide #1574: 2.0 IDE does not come with revisions.txt (release notes)

bperrybap, 2022-10-24:
### Describe the problem

The 2.0 IDE is not released with any sort of release notes.
You can see a release date in the help/about dialog but there does not appear to be any release notes for the release(s).

All 1.x IDE releases came with a file called revisions.txt which was an on going set of release notes.
This was very useful to be able to see when when each release was done and what changes/updates were in each release.
There is even a link to the 1.x release notes on the s/w download page here:
https://www.arduino.cc/en/software
which links to the 1.x revisions.txt file here:
https://www.arduino.cc/en/software/ReleaseNotes

It would be nice & helpful to have that same type of thing for the 2.x releases.
i.e. a revisions.txt file in the zip image, and a link to it on the s/w downloads page so that it can be looked at prior to downloading, installing/extraction.

### To reproduce

Go to the s/w download page:
https://www.arduino.cc/en/software
There is no "Release Notes" link in the "DOWNLOAD OPTIONS" box for the 2.0.0 IDE
(The Legacy 1.8.X release below has this)

Download the linux zip image
Extract the zip image
There is no revisions.txt file in the root directory where the zip image was extracted.

### Expected behavior

A link to release notes on the s/w download page:
https://www.arduino.cc/en/software
in the "DOWNLOAD OPTIONS" box for the 2.0 release

A file called revisions.txt in the root directory where the IDE zip image was extracted.

### Arduino IDE version

2.0.0 zip image

### Operating system

Linux

### Operating system version

Linux Mint 21 MATE

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

bperrybap, 2022-11-18:
Wait...
You guys renamed the title?
changing "does not come with" release notes to "not easily accessible.
So if release notes exist, where are they?

per1234, 2022-11-18:
The reason I renamed it is because you reported two separate issues:

- A release notes file is not included in the Arduino IDE installation.
- The ["Software" page](https://www.arduino.cc/en/software/) does not contain a link to the release notes.

We are declining your request to provide a release notes file with the Arduino IDE installation because such a file is not of any value to the average user (yes, I realize it is valuable to you personally), so the remaining issue is the need to make the release notes more easily accessible. That is definitely something that should be done, and will be simple to accomplish. Thanks for pointing that out.

>So if release notes exist, where are they?

Here:
https://github.com/arduino/arduino-ide/releases

bperrybap, 2022-11-18:
Ah, nice. I hadn't seen that page.
Would be nice to have a link to that on the "Software Page" added into the "Download Options" box
like the one in the 1.x section.

And then what about adding a revision.txt file to the build that is something like

> For release notes see here:
> https://github.com/arduino/arduino-ide/releases

Quick/simple to do and doesn't have to be updated for each release.
This would allow users that are looking for release note find the link from just looking at the installed files.

bperrybap, 2022-12-05:
When fixing this, One thing that would be useful is if there was a single link to a page that could have links to release notes for all the revisions.
Kind of like what Firefox does:
https://www.mozilla.org/en-US/firefox/releases/

Not only is that very convenient for users ( to have access to release notes for all releases), but it also eliminates one more thing/link that would have to be updated on the software download page each time a release is done. 
i.e. having a single link for all releases release notes can eliminate one source of possible human error when doing a release.

closed 2023-01-15 by per1234

bperrybap, 2023-01-15:
awesome!