arduino/arduino-ide #1582: Unable to start debugger when watch expressions are configured

per1234, 2022-10-27:
### Describe the problem

Arduino IDE has [an integrated debugger for debugging sketches](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-debugger).

One of the features of the debugger is the ability to configure expressions to "watch", which is useful for tracking the value of variables at each point in the program execution.

🐛 If a watch expression is configured, the debugger fails to start.

### To reproduce

#### Equipment

Any board of a platform that is configured for use with the debugger (currently only ["**Arduino SAMD (32-bits ARM Cortex-M0+) Boards**" platform](https://github.com/arduino/ArduinoCore-samd) AFAIK).

A CMSIS-DAP compliant or J-Link debug probe. Note that the [Arduino Zero](https://store.arduino.cc/products/arduino-zero) has an on-board debug probe.

#### Steps

1. Set up your board as needed for use with the debugger.
1. Create and save a sketch that contains a variable you can watch:
   ```cpp
   int foo;
   void setup() {}
   void loop() {
     foo++;
   }
   ```
1. If you are using a J-Link, add the `debug_custom.json` file to the sketch:
   https://docs.arduino.cc/tutorials/mkr-wifi-1010/mkr-jlink-setup#debugging-in-arduino-ide-20
1. Enable **Sketch > Optimize for Debugging** in the Arduino IDE menus.
1. Select **Sketch > Upload** from the Arduino IDE menus.
1. Wait for the upload to finish.
1. Click the "**Start Debugging**" button in the IDE toolbar.
1. Wait for the debugger to start.
1. Click the **❯** control next to "**WATCH**" in the "**DEBUG**" view to expand the "**WATCH**" section if it is not already expanded.
1. Click the "**+**" icon ("**Add Expression**") button in the "**WATCH**" section heading.
1. In the "**Edit Watch Expression**" field, type `foo`
1. Click the <kbd>**OK**</kbd> button.
   🙂 The `foo` variable is watched as expected.
1. Click the red square icon ("**Stop**") on the debugger toolbar.
1. Click the green triangle icon ("**Start Debugging**") on the debugger toolbar.

🐛 The debugger does not start. An error notification is shown:

> **Request 4 cancelled on connection close**

It will no longer be possible to start the debugger for that sketch even after restarting the IDE.

The debugger can only be started once more after removing the watch expression.

### Expected behavior

Debugger can be started even if a watch expression is configured.

### Arduino IDE version

2.0.1-snapshot-2b2463b

### Operating system

- Windows
- Linux

### Operating system version

- Windows 10
- Ubuntu 20.04

### Additional context

[... 71 lines cut ...]
    at _.evaluateRequest (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:39565)
    at _.dispatchRequest (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:1:12109)
    at _.handleMessage (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:42356)
root ERROR [hosted-plugin: 14104] (node:10576) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
(node:10576) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
```

---

It may be related to https://github.com/arduino/arduino-ide/issues/412 in that this is a report of the watch expression not working if it was configured before the debugger was started.

---

Originally reported at:

- https://forum.arduino.cc/t/getting-the-debugger-to-work/1020797/1
- https://forum.arduino.cc/t/getting-the-debugger-to-work/1020797/8
- https://forum.arduino.cc/t/arduino-zero-debug-error-with-ide-2-0/1046036/1
- https://forum.arduino.cc/t/arduino-zero-debug-error-with-ide-2-0/1046036/4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-10-28:
This change is suspicious: https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b#diff-0b4e3216be464eb4dd98e2e90c65bdd93b023735c8125d201881d3be43eff3e4

kittaakos, 2022-10-28:
> he problem still occurs, except without the `ENOENT` error in the logs, after I create `c:\Users\per\Documents\Arduino\sketch_oct26b\.vscode\.cortex-debug.registers.state.json`, which makes me question whether that part of the output is relevant.
> 
> Prior to [112153f](https://github.com/arduino/arduino-ide/commit/112153fb965f63d952d126c8244cd3f84f0a1a1b), there was similar output:
> 
> ```
> root INFO [hosted-plugin: 14104] PLUGIN_HOST(14104): PluginManagerExtImpl/loadPlugin(E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\extension)
> root INFO [hosted-plugin: 14104] Debug configuration provider has been registered: cortex-debug, trigger: 1
> root ERROR [hosted-plugin: 14104] (node:10576) [DEP0005] DeprecationWarning: Buffer() is deprecated due to security and usability issues. Please use the Buffer.alloc(), Buffer.allocUnsafe(), or Buffer.from() methods instead.
> root ERROR [hosted-plugin: 14104] (node:10576) UnhandledPromiseRejectionWarning: TypeError: Cannot read property 'showDevDebugOutput' of undefined
>     at _.wrapTimeStamp (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:21659)
>     at _.handleMsg (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:22029)
>     at _.<anonymous> (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:39946)
>     at Generator.next (<anonymous>)
>     at E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:4272
>     at new Promise (<anonymous>)
>     at s (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:4017)
>     at _.evaluateRequest (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:39565)
>     at _.dispatchRequest (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:1:12109)
>     at _.handleMessage (E:\ide2\tip\1-69ac1f4\resources\app\plugins\cortex-debug\extension\dist\debugadapter.js:9:42356)
> root ERROR [hosted-plugin: 14104] (node:10576) UnhandledPromiseRejectionWarning: Unhandled promise rejection. This error originated either by throwing inside of an async function without a catch block, or by rejecting a promise which was not handled with .catch(). (rejection id: 1)
> (node:10576) [DEP0018] DeprecationWarning: Unhandled promise rejections are deprecated. In the future, promise rejections that are not handled will terminate the Node.js process with a non-zero exit code.
> ```

I remember having this error from day one. I downloaded `beta-1`, and here is the same error. It never worked.

```
Version: 2.0.0-beta.1
Date: 2021-02-12T12:10:18.013Z
CLI Version: nightly-20210212 alpha [560025a6]
```

```
root ERROR [hosted-plugin: 75334] Error: ENOENT: no such file or directory, open '/Users/a.kitta/Documents/Arduino/debug/.vscode/.cortex-debug.registers.state.json'
    at Object.openSync (fs.js:440:3)
    at Object.func [as openSync] (electron/js2c/asar.js:140:31)
    at Object.writeFileSync (fs.js:1265:35)
    at t.RegisterTreeProvider._saveState (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/dist/extension.js:212:12839)
    at t.RegisterTreeProvider.debugSessionTerminated (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/dist/extension.js:212:13109)
    at k.debugSessionTerminated (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/plugins/cortex-debug/extension/dist/extension.js:127:49212)
    at /Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/node_modules/@theia/core/lib/common/event.js:176:33
    at CallbackList.invoke (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/node_modules/@theia/core/lib/common/event.js:191:39)
    at Emitter.fire (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/node_modules/@theia/core/lib/common/event.js:318:29)
    at DebugExtImpl.<anonymous> (/Users/a.kitta/Desktop/ide beta-1/Arduino IDE.app/Contents/Resources/app/node_modules/@theia/plugin-ext/lib/plugin/node/debug/debug.js:315:60) {
  errno: -2,
  syscall: 'open',
  code: 'ENOENT',
  path: '/Users/a.kitta/Documents/Arduino/debug/.vscode/.cortex-debug.registers.state.json'
}

```

DeqingSun, 2022-11-05:
@per1234 
Thank your for mentioning the bug in the debug issue. Yes I found the one sketch that has the error and there was a watch item.

Once removed, the error disaapears. Once added back, error appears again.

commit 5d294432 (2022-11-28): feat: patched the Theia debug functionality

Patch for:
 - eclipse-theia/theia#11871
 - eclipse-theia/theia#11879
 - eclipse-theia/theia#11880
 - eclipse-theia/theia#11885
 - eclipse-theia/theia#11886
 - eclipse-theia/theia#11916

Closes #1582
commit 8e5a9a2a (2022-11-29): feat: patched the Theia debug functionality

Patch for:
 - eclipse-theia/theia#11871
 - eclipse-theia/theia#11879
 - eclipse-theia/theia#11880
 - eclipse-theia/theia#11885
 - eclipse-theia/theia#11886
 - eclipse-theia/theia#11916

Closes #1582
commit c217bed6 (2022-11-29): feat: patched the Theia debug functionality

Patch for:
 - eclipse-theia/theia#11871
 - eclipse-theia/theia#11879
 - eclipse-theia/theia#11880
 - eclipse-theia/theia#11885
 - eclipse-theia/theia#11886
 - eclipse-theia/theia#11916

Closes #1582
PR #1706 (2022-12-01): feat: Updated to `cortex-debug@1.5.1`
~**Depends on #1662**~ ✅ 

----

### Motivation
To use a more recent version of the `cortex-debug` VS Code extension. 

### Change description

### Other information

Closes #246
Closes #1582

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixes / features)
closed 2022-12-01 by kittaakos