arduino/arduino-ide #1587: Connection got disposed

ednieuw, 2022-10-27:
### Describe the problem

When compiling the compiler display the error: Connection got Disposed

### To reproduce

Compile the program below and it will stop here:
Using library FS at version 2.0.0 in folder: C:\Users\ednie\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.5\libraries\FS 

char BLEbroadcastName[] ="WordClockV040";

#include <NimBLEDevice.h>       // For BLE communication  https://github.com/h2zero/NimBLE-Arduino
#include <EEPROM.h>            // Used for EEPROM storage of the word clock settings 
#include <ESPNtpClient.h>      // https://github.com/gmag11/ESPNtpClient
#include <WiFi.h>              // Used for NTP time and web page
#include <AsyncTCP.h>          // Used for webpage   https://github.com/me-no-dev/ESPAsyncWebServer
#include <ESPAsyncWebServer.h> // Used for webpage   https://github.com/me-no-dev/ESPAsyncWebServer

### Expected behavior

No errors

### Arduino IDE version

V 2.0.1

### Operating system

Windows

### Operating system version

w11

### Additional context

The program compiles with IDE 1.8.19
Same problem was with V2.0.0
Output compiler in ZIP file
Latest libraries (okt 2022)
Compiles for ESP32-C3 
I did not verify with the nightly build

File SAVEAS is still shitty 
[Disposed.zip](https://github.com/arduino/arduino-ide/files/9882373/Disposed.zip)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-28:
Hi @ednieuw. Thanks for your report.

I'm going to ask you to provide some additional information that might help us to identify the problem.

Please do this:

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Start Windows "**File Explorer**".
1. Open the Arduino IDE installation folder
   (usually `C:\Program Files\Arduino IDE` or `C:\Users\<user name>\AppData\Local\Programs\Arduino IDE`).
1. In the folder listing of the Arduino IDE installation folder, hold the **<kbd>Shift</kbd>** key while clicking the right hand button on the mouse.
1. From the context menu, click "**Open PowerShell window here**".
   PowerShell will now open with the current directory set to the Arduino IDE installation folder.
1. Type the following command:
   ```text
   & ".\Arduino IDE"
   ```
1. Press the **<kbd>Enter</kbd>** key.
   Debug output should now be printed to the PowerShell window as the Arduino IDE starts.
1. Wait for the Arduino IDE to finish starting.
1. Repeat the process you followed before to produce this "**Connection got disposed.**" error.
1. Switch back to the PowerShell window.
1. Right click on the title bar of the PowerShell window.
   This will open a context menu.
1. Select **Edit > Select All** from the context menu.
   This will select all the text in the PowerShell window.
1. Press **<kbd>Ctrl</kbd>+<kbd>C</kbd>**.
   This will copy the contents of the PowerShell window to the clipboard.
1. Paste (**<kbd>Ctrl</kbd>+<kbd>V</kbd>**) the output in a reply to this GitHub issue thread.

---

Please let me know if you have any questions or problems while following those instructions.

ednieuw, 2022-10-28:
PS C:\Program Files\Arduino IDE> & ".\Arduino IDE"
PS C:\Program Files\Arduino IDE>
Arduino IDE 2.0.1
Starting backend process. PID: 9780
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///c%3A/Users/ednie/.theia'
root INFO Backend Object.initialize: 3.9 ms [Finished 0.913 s after backend start]
root INFO Backend Object.configure: 3.1 ms [Finished 0.913 s after backend start]
root INFO Backend MessagingContribution.onStart: 1.5 ms [Finished 0.913 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 0.3 ms [Finished 0.942 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 0.3 ms [Finished 0.942 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.1 ms [Finished 0.942 s after backend start]
root INFO Backend LocalizationBackendContribution.initialize: 29.0 ms [Finished 0.942 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 0.2 ms [Finished 0.942 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 1.3 ms [Finished 0.944 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.5 ms [Finished 0.944 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 0.9 ms [Finished 0.944 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 0.4 ms [Finished 0.946 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 1.3 ms [Finished 0.947 s after backend start]
root INFO Backend NodeFileUploadService.configure: 2.4 ms [Finished 0.947 s after backend start]
daemon INFO Starting daemon from C:\Program Files\Arduino IDE\resources\app\node_modules\arduino-ide-extension\build\arduino-cli.exe...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 0.8 ms [Finished 0.948 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 0.5 ms [Finished 0.948 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.4 ms [Finished 0.949 s after backend start]
root INFO Backend PluginApiContribution.configure: 0.5 ms [Finished 0.950 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.4 ms [Finished 0.951 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.0 ms [Finished 0.951 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 0.6 ms [Finished 0.952 s after backend start]
root INFO Theia app listening on http://localhost:65347.
root INFO Finished starting backend application: 5.4 ms [Finished 0.956 s after backend start]
root WARN The local plugin referenced by local-dir:/c%3A/Users/ednie/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/c%3A/Users/ednie/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:C:\Users\ednie\.arduinoProIDE\plugins does not exist.
Restoring workspace roots: c:\Users\ednie\Documents\Files\Arduino\Pocuter\PocuterWordClockV040
isTempSketch: false. Input was c:\Users\ednie\Documents\Files\Arduino\Pocuter\PocuterWordClockV040\PocuterWordClockV040.ino
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-hu" to a VS Code extension "vscode-language-pack-hu@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
[... cut ...]
INFO[0101] Required tool                                 tool="esp32:riscv32-esp-elf-gcc@gcc8_4_0-esp-2021r2-patch3"
INFO[0101] Required tool                                 tool="esp32:xtensa-esp32-elf-gcc@gcc8_4_0-esp-2021r2-patch3"
INFO[0101] Required tool                                 tool="esp32:xtensa-esp32s2-elf-gcc@gcc8_4_0-esp-2021r2-patch3"

daemon INFO INFO[0101] Required tool                                 tool="esp32:xtensa-esp32s3-elf-gcc@gcc8_4_0-esp-2021r2-patch3"

[10728:1028/124413.983:ERROR:gpu_init.cc(453)] Passthrough is not supported, GL is disabled, ANGLE is