arduino/arduino-ide #1590: Additional sketches can't be opened via file association/command line argument

per1234, 2022-10-28:
### Describe the problem

In addition to opening them via the Arduino IDE GUI, sketches can be opened directly either via file association or by passing their path to the IDE command line invocation.

🐛 This only works once per IDE session. After that, attempting to open additional sketches in this manner has no effect.

### To reproduce

1. Select **File > Quit** from the Arduino IDE menus if it is running.
1. Open a sketch via file association or command line argument:
   ```text
   & ".\Arduino IDE" "C:\Users\per\Documents\Arduino\Foo"
   ```
   🙂 The IDE starts up with the specified sketch open, as expected.
1. Open a different sketch via file association or command line argument:
   ```text
   & ".\Arduino IDE" "C:\Users\per\Documents\Arduino\Bar"
   ```

🐛 The command exits immediately with no effect. The specified sketch is not opened in an IDE window.

### Expected behavior

Additional sketches can be opened via file association or command line argument when the IDE is already running.

### Arduino IDE version

2.0.1

### Operating system

- Windows
- Linux

### Operating system version

- Windows 10
- Ubuntu 20.04

### Additional context

I bisected the regression to 2b2463b / https://github.com/arduino/arduino-ide/pull/1563

---

Originally reported at https://forum.arduino.cc/t/arduino-ide-2-0-1-is-now-available/1046764/7

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-11-02:
> This is on macOS

https://github.com/arduino/arduino-ide/blob/f6275f9f62e66df091ac6e12bfc78d56f36c2380/arduino-ide-extension/src/electron-main/theia/electron-main-application.ts#L359

closed 2022-11-10 by kittaakos

dalnel, 2025-05-03:
2.3.5 them same question

PR #1617 (2022-11-10): fix: avoid ENOTDIR when opening second instance. 
### Motivation

To allow opening a second sketch from the command line.

### Change description

 - Fix incorrect logic when detecting if the arg is a valid sketch,
 - Allow opening a second instance on macOS.

### Other information

Closes #1590

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixe [... cut]
commit c3dc7c63 (2022-11-10): fix: avoid ENOTDIR when opening second instance.

If the resource is a file, do not try to `readdir`, but return undefined

Closes #1590