arduino/arduino-ide #1597: Serial plotter does not plot

slavicd, 2022-10-29:
### Describe the problem

Latest IDE (2.0.2) serial plotter is empty. No value gets plotted and 

### To reproduce

Install a fresh nightly build and try to plot some values:

```cpp
Serial.print("P: ");
Serial.println(cfPitch);
```

### Expected behavior

Plotter shows a plot of values.

### Arduino IDE version

2.0.2

### Operating system

Linux

### Operating system version

Fedora 36

### Additional context

Wayland 42.4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

nmzaheer, 2022-10-29:
Can you share the entire code to reproduce this ?

slavicd, 2022-10-29:
Here is an MVE:

```cpp
void mySetup() 
{
    pinMode(A1, INPUT);
    Serial.begin(115200);
    delay(100);
}

int main(void) 
{
    init();
    mySetup();

    while(1) {
        Serial.print("A1: ");
        Serial.println(analogRead(A1));
        delay(300);
    }

    return 0;
}

```

slavicd, 2022-10-29:
Apparently it's (the plotter) thrown off by the "A1:" label. I can not reproduce without the label. This used to work just fine, though in 2.0. Is there a new way of printing labels, perhaps?

nmzaheer, 2022-10-29:
I've had a look at your code. You will have to remove the space after the label i.e change it to `Serial.print("A1:");`.

Please have a look at the [Serial Protocol](https://github.com/arduino/Arduino/blob/master/build/shared/ArduinoSerialPlotterProtocol.md).  The **Label and Value message** format does not have a space after the colon `:`.

slavicd, 2022-10-29:
This is one of those cases where the protocol was "reverse engineered" and it so happens it used to work with a space before 2.0.1 . Thanks for the doc.

closed 2022-10-29 by slavicd