arduino/arduino-ide #1599: Creation of sketches with non-compliant names is permitted

per1234, 2022-10-29:
Creation of sketches with non-compliant names is permitted
### Describe the problem

The Arduino sketch specification defines the characters allowed for use in sketch folder and file names:

https://arduino.github.io/arduino-cli/dev/sketch-specification/#sketch-folders-and-files

🐛 Arduino IDE allows users to create sketches with non-compliant names

### To reproduce

1. Select **File > New** from the Arduino IDE menus.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch under the sketchbook folder, with the name `` `~!@#$%^&()=+[{]};', ``
   🐛 The IDE allowed the creation of a sketch with non-compliant name.
1. Click the **●●●** icon on the right side of the editor toolbar.
1. Select "**New Tab**" from the menu.
1. In the "**Name for new file**" field, type `` `~!@#$%^&()=+[{]};',2 ``
1. Click the <kbd>**OK**</kbd> button.
   🐛 The IDE allowed the creation of an additional sketch file with non-compliant name.
1. Select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
   🐛 Compilation fails with a cryptic error:
   ```text
   avr-g++: error: C:\Users\per\AppData\Local\Temp\arduino-sketch-DC0B853C212DB0BA0ED269B37923BDC7\sketch\`~!@#$%^&()=+[;',.ino.cpp: No such file or directory
   avr-g++: warning: '-x c++' after last input file has no effect
   avr-g++: fatal error: no input files
   compilation terminated.
   
   exit status 1
   
   Compilation error: exit status 1
   ```
1. Start Arduino IDE 1.x
   🐛 A warning dialog is shown on startup:
   TODO
   🐛 The sketch can not be shared with Arduino IDE 1.x users due to the non-compliant name.
   This problem seems less significant in this contrived demo where the sketch name causes compilation failure, but other non-compliant sketch names will not have an obvious impact to the Arduino IDE 2.x user.
1. Import the sketch to Arduino Web Editor
   ⚠ Due to a bug in Arduino Web Editor, the sketch will be impossible to delete and might possibly cause other problems, so I don't recommend actually doing this with any Arduino account you care about.
   🐛 There is an error message on import.
1. Delete the imported sketch from your Arduino Web Editor sketchbook.
   🐛 The sketch is impossible to delete:
   > **Error while deleting `~!@#$%^&()=+[{]};',**

🐛 Arduino IDE 2.x allows the creation of sketches that are either unusable or not portable to other official development software.

### Expected behavior

Arduino IDE 2.x does not allow the creation of sketches with non-compliant names.

Sketches created with one development tool are usable in all other tools.

### Arduino IDE version

2.0.1

### Operating system

- Windows
- Linux

### Operating system version

- Windows 10
- Ubuntu 20.04

### Additional context

Originally reported by @UKHeliBob at https://forum.arduino.cc/t/file-save-as-allows-invalid-filenames-to-be-used/700966

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2023-01-16:
I have collected the differences between the spec and how it works in the create editor https://create.arduino.cc/editor:

Allowed to start with:
 - spec   -> `[0-9a-zA-Z]`
 - create -> `[0-9a-zA-Z_]`

Allowed characters:
 - spec   -> `[0-9a-zA-Z_\.-]`
 - create -> `[0-9a-zA-Z]`

Allowed length:
 - spec   -> `{1-63}`
 - create -> `{1-36}`

The problem is that the [Create API](https://api2.arduino.cc/create/docs#) is more restrictive than the spec. Hence, it won't be possible in IDE2 to constantly push any local sketch to the cloud due to the invalid name. To summarize, what is valid and works as a local sketch might not be OK as a remote sketch. I need to know how this was designed to handle in IDE2. Thanks!

kittaakos, 2023-01-16:
As a proposal, IDE2 can use the spec rules when creating ordinary sketches, and when the user wants to push them, and the name of the sketch does not comply with the Create API naming convention, users must rename it before a push. What do you think, @91volt?

kittaakos, 2023-01-17:
> Sketches created with one development tool are usable in all other tools.

I do not think this is always feasible. Example: IDE2 has a valid sketch name, export the sketch to ZIP, still valid, and import it to Create editor, it could be invalid.

per1234, 2023-01-18:
It is absolutely feasible. The Arduino Cloud developers just need to respect the Arduino Sketch Specification.

kittaakos, 2023-01-24:
> Expected behavior

IDE2 has to validate the [sketch code files](https://arduino.github.io/arduino-cli/dev/sketch-specification/#additional-code-files) besides the main sketch file and the sketch folder. Validation of any new file is trivial; the question was when to validate the sketch code files of existing sketches, how to warn users, and how to guide them to fix the issues. Per and I further refined the requirements of this task and decided to do the following:

 - IDE2 will show any sketches under `File` > `Sketchbook` if the sketch folder name matches the main sketch file name without the extension. This is sufficient for IDE2 to ask the CLI to [load the sketch](https://arduino.github.io/arduino-cli/dev/rpc/commands/#loadsketchrequest).
 - Unlike IDE 1.x, IDE2 won't validate other sketches in the sketchbook on app start, only the sketch being opened.
 - IDE2 will propose a series of actions to fix the problem when a user opens an invalid sketch. IDE2 can prompt the following actions. The order of the actions matter:
    - Rename the sketch folder and the main sketch file if invalid (this should be handled with a `Save as...`, after fixing the folder and main sketch file name, IDE2 deletes the previous sketch and reloads the renamed sketch in the same window,
    - The same validation logic runs. The sketch folder name must be valid at this point. IDE2 will prompt actions to fix any code file naming issues (IDE2 will use the new file, rename file dialog)
    - Any time the user interrupts the series sketch folder or code file renaming, IDE2 will interrupt the sketch opening and opens a new temp sketch in the same window.
    - The series of proposed actions happen in a modal fashion. IDE2 does not allow users to access the UI (compile or upload sketch, change boards, etc.) until the sketch is valid.

commit 60a64b1c (2023-01-24): fix: enforce valid sketch folder name on `Save as`

Closes #1599
commit 2aea3df7 (2023-01-25): fix: enforce valid sketch folder name on `Save as`

Closes #1599
commit 84e9ea3f (2023-01-31): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
commit b021d3b9 (2023-02-01): feat: rename, deletion, and validation support

Closes #1599
Closes #1825
Closes #1826
closed 2023-02-15 by kittaakos