arduino/arduino-ide #1600: "Sketchbook" menu missing if sketch name is invalid regex

per1234, 2022-10-29:
### Describe the problem

The **File > Sketchbook** menu provides convenient access to the sketches the user has stored under their sketchbook folder.

During the population of this menu, the IDE processes each sketch name as a regular expression:

https://github.com/arduino/arduino-ide/blob/0d0550974abd26adffd53a9d3036172705dd5177/arduino-ide-extension/src/node/sketches-service-impl.ts#L160

🐛 If a sketch with a name that happens to be an invalid regular expression is present in the sketchbook, the **File > Sketchbook** menu is missing.

### To reproduce

1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch under the sketchbook folder with the name `Foo`
1. Select **File > Sketchbook** from the Arduino IDE menus.
   🙂 The "Foo" sketch and any other sketches present in the sketchbook are listed in the menu.
1. Select **File > Save As...** from the Arduino IDE menus.
1. Save the sketch under the sketchbook folder with the name `Bar++`
1. Select **File** from the Arduino IDE menus.

🐛 There is no **File > Sketchbook** menu.

### Expected behavior

**File > Sketchbook** menu is populated with all valid sketches even if invalid sketches are also present in the sketchbook.

### Arduino IDE version

2.0.1

### Operating system

- Windows
- Linux

### Operating system version

- Windows 10
- Ubuntu 20.04

### Additional context

This issue is not about supporting non-compliant sketch names. I used **File > Save As...** in the demo in order to make it as simple as possible to reproduce this bug. However, even after the bug of the IDE allowing the creation of non-compliant sketch names (https://github.com/arduino/arduino-ide/issues/1599) is fixed, the user may still create or find such sketches (which will become increasingly prevalent since the IDE allows their creation) in the wild. The IDE must handle the presence of such invalid sketches gracefully, rather than breaking.

---

One option for resolving the issue would be to escape the `sketchName` string for use in a regular expression, as done here:

https://stackoverflow.com/a/6969486/7059512

---

I bisected the introduction of the bug to 4c55807 / https://github.com/arduino/arduino-ide/pull/1018 (does not occur with the build for cb50d3a)

---

Originally reported by @Willem43T at https://forum.arduino.cc/t/2-0-1-no-open-recent-menu-item/1046905/6

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

ubidefeo, 2022-11-09:
This issue is also present when a fresh installation of IDE 2 is first launched.
The `Documents/Arduino` folder is created but being empty it won't populate the menu item, which, being long 0, is just hidden

kittaakos, 2022-11-09:
> The `Documents/Arduino` folder is created but being empty it won't populate the menu item, which, being long 0, is just hidden

The issue you are referring to is already logged here, @ubidefeo  https://github.com/arduino/arduino-ide/issues/569

I am going to revert your assigned changes. There is a PR https://github.com/arduino/arduino-ide/pull/1622 for this task.

commit d304922b (2022-11-07): fix: escaped regex chars in pattern

Closes #1600
commit 88e3de6f (2022-11-10): fix: escaped regex chars in pattern

Closes #1600
PR #1622 (2022-11-10): fix: escaped regex chars in pattern
### Motivation

If regex patterns are not correctly escaped, they will break. Such as the `File` > `Sketchbook` issue when there is a sketch with the following invalid name: `bar++`.

### Change description

 - Reuse the regex pattern escape mechanism from VS Code.
 - Remove strict [`is-valid-path`](https://www.npmjs.com/package/is-valid-path) check. Otherwise, on Windows, it's impossible to open a sketch in the following way: `"Arduino IDE.exe" "..\\..\\Documents\\Arduino\\bar++"` input will give a false negative result.

### Other information

Closes #1600 [... cut]
commit f6d112e1 (2022-11-10): fix: escaped regex chars in pattern

Closes #1600
closed 2022-11-10 by kittaakos