arduino/arduino-ide #1601: Sometimes it will not compile after switching development version

Tinyu-Zhao, 2022-10-30:
### Describe the problem

Sometimes it fails to compile after switching to the development version, and it says "No connection established".

[image]

But strangely enough, after restarting the IDE, most of them will be able to compile normally.

### To reproduce

Sometimes switching clients and compiling will cause this problem, and sometimes not.

### Expected behavior

Normal compilation

### Arduino IDE version

Normal compilation

### Operating system

macOS

### Operating system version

Big Sur 11.7 (20G817)

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

Tinyu-Zhao, 2022-11-03:
Please help me to look at this problem @per1234

per1234, 2022-11-03:
I have seen a recent spike in reports of compilation failing with this "No connection established" error.

It started soon after the 2.0.1 release, so the spike might be correlated to people updating from 2.0.0 to 2.0.1.

This is a fairly generic error, so I'm not certain all the reports have the same cause:

- https://forum.arduino.cc/t/linux-nano-ch340-detected-working-in-ide1-8-13-but-not-ide2-0-1/1048313/10
- https://forum.arduino.cc/t/no-connections-established/1048343/1
- https://forum.arduino.cc/t/no-connections-established/1048343/10
- https://forum.arduino.cc/t/no-connections-established/1048343/13
- https://forum.arduino.cc/t/2-0-1-gets-confused-when-making-new-project/1048866/1
- https://forum.arduino.cc/t/2-0-1-gets-confused-when-making-new-project/1048866/8
- https://forum.arduino.cc/t/aitrip-esp32-dev-kit-v1-board-not-connecting-to-arduino-ide/1048870

per1234, 2022-11-03:
Hi @Tinyu-Zhao. Thanks for your report.

In order to gather more information that might help us to troubleshoot your problem, I'm going to ask you to run the Arduino IDE from the command line, then share the output.

Please do this:

1. Open your **Applications** folder.
1. Open **Utilities**.
1. Double-click on "**Terminal**".
1. Type the following command:
   ```text
   "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
   ```
1. Press **<kbd>Enter</kbd>**
1. Take whatever actions are required to produce the "No connection established" error.
1. Switch back to the terminal window if the Arduino IDE window took the focus.
1. Press **<kbd>⌘</kbd>+<kbd>A</kbd>**.
   This will select all the text in the terminal window.
1. Press **<kbd>⌘</kbd>+<kbd>C</kbd>**.
   This will [copy](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the selected text to the clipboard.
1. Click on the "**Leave a comment**" field here on this GitHub issue.
1. Press **<kbd>⌘</kbd>+<kbd>V</kbd>**.
   This will [paste](https://lifehacker.com/how-to-copy-cut-and-paste-for-beginners-5801525) the output into the code block.
1. Click the <kbd>**Comment**</kbd> button to make the reply.

---

Please let me know if you have any questions or problems while following those instructions.

---

> after switching to the development version

I'm not sure I understood correctly what you mean by this. Please provide a more detailed description to help me to understand it.

kldorseyNU, 2022-11-08:
Hi @per1234. I'm having the same issue. If I try to compile the Blink example for an Arduino AVR, (e.g., Uno), it compiles correctly. Switching to the ESP32 board, which is my goal, creates the no connection established error even if I'm purely trying to compile. Switching back to Uno also creates the error. Attached is my terminal log showing successful compile to Uno, unsuccessful compile to the ESP32, and following unsuccessful compile to Uno. Thank you for your guidance and advice!
[debug_board_manager.txt](https://github.com/arduino/arduino-ide/files/9962300/debug_board_manager.txt)

numaga94, 2022-11-09:
Last login: Wed Nov  9 18:51:12 on console
➜  ~ "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
➜  ~ "/Applications/Arduino IDE.app/Contents/MacOS/Arduino IDE"
Arduino IDE 2.0.1
Starting backend process. PID: 1881
Using browser-only version of superagent in non-browser environment
Configuring to accept webviews on '^.+\.webview\..+$' hostname.
root INFO Configuration directory URI: 'file:///Users/yu/.theia'
root INFO Backend Object.initialize: 9.2 ms [Finished 1.818 s after backend start]
root INFO Backend Object.configure: 6.2 ms [Finished 1.818 s after backend start]
root INFO Backend MessagingContribution.onStart: 3.0 ms [Finished 1.818 s after backend start]
root INFO Backend OSBackendApplicationContribution.configure: 1.6 ms [Finished 1.876 s after backend start]
root INFO Backend DefaultWorkspaceServer.onStart: 1.0 ms [Finished 1.876 s after backend start]
root INFO Backend LocalizationBackendContribution.configure: 0.3 ms [Finished 1.876 s after backend start]
root WARN Backend LocalizationBackendContribution.initialize took longer than the expected maximum 50 milliseconds: 59.5 ms [Finished 1.878 s after backend start]
root INFO Backend TaskBackendApplicationContribution.onStart: 1.8 ms [Finished 1.878 s after backend start]
root INFO Backend ElectronTokenBackendContribution.configure: 3.9 ms [Finished 1.882 s after backend start]
root INFO Backend PluginDeployerContribution.initialize: 0.8 ms [Finished 1.883 s after backend start]
root INFO Backend ConfigServiceImpl.onStart: 2.1 ms [Finished 1.885 s after backend start]
root INFO Backend WebviewBackendSecurityWarnings.initialize: 3.6 ms [Finished 1.892 s after backend start]
root INFO Backend ArduinoDaemonImpl.onStart: 5.1 ms [Finished 1.894 s after backend start]
root INFO Backend NodeFileUploadService.configure: 9.3 ms [Finished 1.894 s after backend start]
daemon INFO Starting daemon from /Applications/Arduino IDE.app/Contents/Resources/app/node_modules/arduino-ide-extension/build/arduino-cli...
discovery-log INFO start
discovery-log INFO start new deferred
root INFO Backend BoardDiscovery.onStart: 2.2 ms [Finished 1.897 s after backend start]
root INFO Backend FileDownloadEndpoint.configure: 1.1 ms [Finished 1.898 s after backend start]
root INFO Backend AuthenticationServiceImpl.onStart: 0.6 ms [Finished 1.900 s after backend start]
root INFO Backend PluginApiContribution.configure: 1.5 ms [Finished 1.901 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.4 ms [Finished 1.903 s after backend start]
root INFO Backend HostedPluginReader.configure: 0.2 ms [Finished 1.904 s after backend start]
root INFO Backend PlotterBackendContribution.configure: 1.0 ms [Finished 1.906 s after backend start]
root INFO Theia app listening on http://localhost:49482.
root INFO Finished starting backend application: 10.7 ms [Finished 1.913 s after backend start]
root WARN The local plugin referenced by local-dir:/Users/yu/.arduinoIDE/plugins does not exist.
root WARN The local plugin referenced by local-dir:/Users/yu/.arduinoIDE/extensions does not exist.
root WARN The local plugin referenced by local-dir:/Users/yu/.arduinoProIDE/plugins does not exist.
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "cortex-debug" to a VS Code extension "cortex-debug@0.3.10" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-arduino-tools" to a VS Code extension "vscode-arduino-tools@0.0.2-beta.5" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-cpp" to a VS Code extension "cpp@1.52.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json" to a VS Code extension "json@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-builtin-json-language-features" to a VS Code extension "json-language-features@1.46.1" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-bg" to a VS Code extension "vscode-language-pack-bg@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-cs" to a VS Code extension "vscode-language-pack-cs@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-de" to a VS Code extension "vscode-language-pack-de@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-es" to a VS Code extension "vscode-language-pack-es@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-fr" to a VS Code extension "vscode-language-pack-fr@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-hu" to a VS Code extension "vscode-language-pack-hu@1.48.3" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-it" to a VS Code extension "vscode-language-pack-it@1.53.2" with engines:
root INFO PluginTheiaDirectoryHandler: accepting plugin with path
root INFO Resolved "vscode-language-pack-ja" to a VS Code extension "vscode-language-pack-ja@1.53.2" with engines:
[... cut ...]
root ERROR Request getBoardDetails failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request search failed with error: 14 UNAVAILABLE: read ECONNRESET
root ERROR Request installed failed with error: 14 UNAVAILABLE: read ECONNRESET
daemon INFO Daemon exited with exit code: 2.
root ERROR Request getBoardDetails failed with error: 14 UNAVAILABLE: No connection established
root INFO Failed to start language server for esp8266:esp8266:d1_mini_pro
root ERROR Have not received the build path from the CLI while running the compilation.

pdcook, 2022-11-09:
Can confirm that reverting to 2.0.0 fixes this issue. I am also using the ESP32.

barc343, 2022-11-09:
Use a legacy version (1.8.X) for me works