arduino/arduino-ide #1602: Scrollbar covers bottom line of Serial Monitor output

per1234, 2022-10-30:
### Describe the problem

The Arduino IDE "**Serial Monitor**" view has a "**Toggle Autoscroll**" control. When this is enabled, the view automatically scrolls down as the number of lines of data exceed the visible height of the view. The bottom line of the Serial Monitor is the most recent output from the Arduino board, and thus often of the most interest to the user.

When a line in the output contains more characters that will fit in the horizontal area of the Serial Monitor output field, a scrollbar is added at the bottom of the field to allow the user to view all the output.

🐛 Autoscroll incompletely scrolls the output field when the horizontal scrollbar is present, resulting in the most recently printed line only being partly visible.

### To reproduce

#### Equipment

- Any Arduino board capable of serial output.

#### Steps

1. Upload a sketch that produces serial output that exceeds the height and width of the "**Serial Monitor**" view.
   For example:
   ```cpp
   int counter;
   void setup() {
     Serial.begin(9600);
   }
   void loop() {
     Serial.print("hello #");
     Serial.print(counter++);
     Serial.println(" 1234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890123456789012345678901234567890");
     delay(500);
   }
   ```
1. Open the "**Serial Monitor**" view.
1. Select "**9600 baud**" from the dropdown baud rate menu at the top right corner of the "**Serial Monitor**" view.
1. Select the "**Toggle Autoscroll**" icon at the right side of the bottom panel toolbar if it is not already.
1. Wait for the serial output to exceed the visible height of the view.

🐛 The most recently printed line is only partly visible:

[image]

This is especially problematic when the line contains data that is important for the user to see at the time it is printed vs a constant data stream where the previous line serves just as well at the moment as the unreadable one.

### Expected behavior

The most recently printed line is always shown above the bottom edge of the Serial Monitor view.

### Arduino IDE version

2.0.1

### Operating system

Windows

### Operating system version

10

### Additional context

In previous versions of Arduino IDE, the issue occurred even when the scrollbar was not present (https://github.com/arduino/arduino-ide/issues/972). That bug was fixed by https://github.com/arduino/arduino-ide/pull/1446, but this variant specific to the presence of the scrollbar was not resolved by that.

---

The line does become fully visible if I manually scroll the Serial Monitor down.

---

Originally reported at https://forum.arduino.cc/t/ide-2-0-1-serial-monitor-line-at-bottom-partially-obscured-when-line-is-long/1047716

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

PaulStoffregen, 2022-12-01:
Isn't this issue now resolved?

per1234, 2022-12-01:
Thanks @PaulStoffregen. I see it was fixed incidentally by https://github.com/arduino/arduino-ide/pull/1662

closed 2022-12-01 by per1234

FitzkeeLab, 2022-12-31:
I am still having this problem in version 2.0.3 (Win 10 x64).

per1234, 2022-12-31:
Hi @FitzkeeLab. I just tried following the instructions to reproduce the bug provided in the "**To reproduce**" section of [the first post of this issue](https://github.com/arduino/arduino-ide/issues/1602#issue-1428988065) and I am not able to reproduce it when using Arduino IDE 2.0.3.

If you follow those exact instructions, does the fault occur for you?

FitzkeeLab, 2023-01-01:
You're correct - when the line is long, and both the horizontal and vertical scrollbars are used, the monitor automatically advances to the last line. However, when the line does not extend off the screen and does not require the horizontal scroll bar, the terminal is always "short" by one line. 

In the example below (I also increased the delay to 2s to give me more time to check), line #16 is displayed, but I can scroll down to reveal line #17 if I click into the serial monitor window. 

[image]

per1234, 2023-01-01:
Thanks for checking @FitzkeeLab. We are tracking that bug here: https://github.com/arduino/arduino-ide/issues/1736