arduino/arduino-ide #1608: 2.0.1 has loweR VERSION ID than 1.8 releases

BitLoose, 2022-10-31:
### Describe the problem

Having upgraded to 2.0.1 from 1.8 my Sketch is throwing an error because it seems the VERSION ID number is lower than the 1.8 release. Here´s the code that throws, which was fine in IDE 1.8:

#if(ARDUINO < 10810)
#error "Unsupported Arduino IDE version, use Arduino IDE 1.8.10 or later from https://www.arduino.cc/en/software"
#endif

### To reproduce

1. Use IDE 2.0.1
2. Put this code in a sketch:-
#if(ARDUINO < 10810)
#error "Unsupported Arduino IDE version, use Arduino IDE 1.8.10 or later from https://www.arduino.cc/en/software"
#endif
3. Verify

### Expected behavior

Should not throw an error unless Arduino have totally changed the VESIONing system

### Arduino IDE version

2.0.1

### Operating system

Windows

### Operating system version

Latest

### Additional context

None

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-10-31:
Hi @BitLoose. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/725 (this version number is actually set for Arduino IDE 2.x by Arduino CLI).

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-10-31 by per1234

lordfitzy, 2022-11-29:
Hi!

I've recently upgraded the Arduino IDE:
    Version: 2.0.2
    Date: 2022-11-17T11:54:56.612Z
    CLI Version: 0.29.0 [76251df9]

When I compile the command line contains this:
    -DARDUINO=10607

and that's what my program sees for ARDUINO.

Is this expected?  Don't know if it matters but I did not uninstall the previous IDE version, 1.8.19.