arduino/arduino-ide #1610: Port selection - replace board name with port name - it is port selection

pyramidak, 2022-11-01:
### Describe the request

When I select port, I want to see in selection the port mqtt name with IP address "pracovna at 192.168.0.108". 
[image]

That's what matters that is what I want to know to know what I will update.

### Describe the current behavior

When I select port, you show me useless board name that is always same. It it port selection, not a board selection.
[image] 
Can you explain me the logic show me the board name in selection that is always the same? Also board name is too long, so even different board name shows also the same first words. Useless and it is main big button selection what really says lot about you.
[image]

### Arduino IDE version

2.0.1

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-01:
Hi @pyramidak. Thanks for your suggestion.

>It it port selection, not a board selection.

It is both a board and a port selection. The board shown in the menu will be selected, just as if you had selected it from the **Tools > Board** menu. Similarly, the port shown in the menu will be selected, just as if you had selected it from the **Tools > Port** menu.

This menu is most useful for the boards that are automatically [identifiable by Arduino IDE from their port properties](https://arduino.github.io/arduino-cli/latest/pluggable-discovery-specification/#board-identification) (e.g., USB VID/PID). In this case, the user will find it convenient to have the IDE automatically make the correct selections from a single click. In the case of boards with a port which does not have identifying properties, you might find it more convenient to use the **Tools** menus just as you would in Arduino IDE 1.x.

pyramidak, 2022-11-01:
I see. So after port selection (the board is selected according to port) the user do not need to see what he selected despite of the port is the important information and the board is extra. All selections normally shows what was selected but not this time.

per1234, 2022-11-01:
I see what you are requesting now. You would like to see the port name in the toolbar. I previously thought you were requesting the removal of the board name from the menu.

pyramidak, 2022-11-01:
Good I didn´t give up. :-) I´m not sure you can show both in complete selection because of space. During the selection the both information are fine, but after selection is complete, the mqtt device name of port is the key to know what I will update. So I want you to to consider to solve this very quickly and easily. Switch places of port and board and all is solved. The port is key information that should be in bold font in first row and the board information is extra that could be on second row.

kittaakos, 2022-11-02:
> Also board name is too long, so even different board name shows also the same first words.

Related conversation: https://github.com/arduino/arduino-ide/pull/1216#issuecomment-1199619995