arduino/arduino-ide #1612: First time load of project with many files very slow with "the editor has not shown up in time" errors

FrightRisk, 2022-11-01:
### Describe the problem

We have a project that has 50+ .cpp and .h files. The first time you load this project, the 2.0 IDE is doing some kind of check on the files and slowly loads them one at a time in the IDE file tabs. This does not occur the next time the project is opened. The load takes several minutes. The second time through, it is still slow, but less than 30 seconds.

[arduino ide boot log 5.txt](https://github.com/arduino/arduino-ide/files/9913865/arduino.ide.boot.log.5.txt)

### To reproduce

Go to this repo, download the project, and try to load it into the 2.0 IDE
https://github.com/DCC-EX/CommandStation-EX

### Expected behavior

I would expect it to load much faster and not have a 5 second timeout for every file.

### Arduino IDE version

2.02 nightly 20221030

### Operating system

Windows

### Operating system version

Windows 11

### Additional context

boot log included above

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

FrightRisk, 2022-11-01:
Can anyone explain what the load process is and why it is so much slower the first time it sees a new project? Once a hash is created for a .ino file, the IDE skips whatever check it does the first time. Also, in order to generate logs, I need it to think the project is one it hasn't seen. I tried deleting the files in C:\users\<user name\.arduino called recent-sketches.json and recentworkspace.json, but that just clears the recent files menu. It is storing something somewhere else that knows that file was opened and processed before.

per1234, 2022-11-01:
Thanks for your report @FrightRisk 

>It is storing something somewhere else that knows that file was opened and processed before.

I already provided you with this information on your forum post:

https://forum.arduino.cc/t/how-to-clear-recent-files-list-in-2-0/1047673/7

If that isn't working for you, or if there is something you didn't understand about the informatoin I shared, please reply on the forum thread and I'll help you out.

kittaakos, 2022-11-07:
It seems it affects all sketches created before `2.0.1` and opened with `2.0.1` for the first time. IDE2 does not have this problem opening the same sketch for the second time.

Can somebody please confirm this?

Here is my test sketch as `Errors.zip`. (Original [source](https://github.com/arduino/arduino-ide/pull/1064#issue-1272314345)):

> * Compile a sketch with multiple errors. Here is an example sketch:
>   [Errors.zip](https://github.com/arduino/arduino-ide/files/8911126/Errors.zip)

 - Unpack the file and put it into your sketchbook (rename the folder and the main sketch file if required),
 - Open IDE2 (it opens with your last sketches or a temp one),
 - Go to `File` > `Sketchbook` and open the test sketch,
 - The first editor opens quickly, then from the second editor, every editor takes 5 sec to open,
 - Quit IDE2,
 - Open IDE2 the test sketch quickly restores. 

👆 It's a bug; no question about it, but please confirm. Thank you!

per1234, 2022-11-07:
I confirm.

If you want an extreme example of this bug, try opening an older version of the popular Marlin 3D printer firmware sketch from before they restructured it in 2.0.0.0 by moving most of the source files under the `src` subfolder:

https://github.com/MarlinFirmware/Marlin/archive/refs/tags/1.1.9.1.zip

That sketch actually brings my computer to its knees and causes other applications to crash due to lack of memory.

commit c87d3bc6 (2022-11-07): fix: flawed timing issue when opening editors

From now on, the editor widget open promise resolution does not rely on
internal Theia events but solely on @phosphor's `isAttached`/`isVisible`
properties.
The editor widget promise resolves with the next task after a navigation
frame so the browser can render the widget.

Closes #1612
commit b6c47774 (2022-11-10): fix: flawed timing issue when opening editors

From now on, the editor widget open promise resolution does not rely on
internal Theia events but solely on @phosphor's `isAttached`/`isVisible`
properties.
The editor widget promise resolves with the next task after a navigation
frame so the browser can render the widget.

Closes #1612
commit cc57dece (2022-11-10): fix: flawed timing issue when opening editors

From now on, the editor widget open promise resolution does not rely on
internal Theia events but solely on @phosphor's `isAttached`/`isVisible`
properties.
The editor widget promise resolves with the next task after a navigation
frame so the browser can render the widget.

Closes #1612
commit 37355530 (2022-11-11): fix: flawed timing issue when opening editors

From now on, the editor widget open promise resolution does not rely on
internal Theia events but solely on @phosphor's `isAttached`/`isVisible`
properties.
The editor widget promise resolves with the next task after a navigation
frame so the browser can render the widget.

Closes #1612
closed 2022-11-11 by kittaakos