arduino/arduino-ide #1614: Strange bug: Lines with more than 486 chars leads to wrong color parsing, Autoformat leads to compile errors

CptHolzschnauz, 2022-11-02:
### Describe the problem

Sketch with a long line leads to strange behaviour in the IDE. 

- Wrong color parsing after col +/- 500
- Compile / Not compile depends on format not on the code itself
- Flaky compile behaviour: Sometimes compiles with wrong color coding sometimes not, i see no rule

In a long line of code, after character +/- 500 the IDE stops coloring:
[image]
but only when this line is a spaced to the right:
[image]

If it's on the same "width", the code from this line on is wrong parsed:
[image]

Sideffect: If i autoformat that sketch it won't compile anymore.

Effect : Sometimes it compiles with that wrong color parsing, sometimes not. 
I think the IDE sends somehow different code to the compiler, depending on its format?

### To reproduce

Put that sketch snippet in the IDE, around col 500 the effect starts:

`void setup() {
  // put your setup code here, to run once:
}

void loop() {
  // put your main code here, to run repeatedly:
 mqtt_status4 = "Online";
          mqtt_msg = "('" + mqtt_device_id + "','" + String(longitude_tft, 10) + "','" + String(latitude_tft, 10) + "','" + mqtt_alarm + "','" + String(B1Volt, 1) + "','" + String(B2Volt, 1) + "','" + Batt_Temp11 + "','" + Batt_Temp12 + "','" + Batt_Temp13 + "','" + Batt_Temp14 + "','" + Batt_Temp15 + "','" + Batt_Temp16 + "','" + Batt_Temp21 + "','" + Batt_Temp22 + "','" + Batt_Temp23 + "','" + Batt_Temp24 + "','" + Batt_Temp25 + "','" + Batt_Temp26 + "','" + Batt_Temp31 + "','" + Batt_Temp32 + "','" + Batt_Temp33 + "','" + Batt_Temp34 + "','" + Batt_Temp35 + "','" + Batt_Temp36 + "','" + Batt_Temp41 + "','" + Batt_Temp42 + "','" + Batt_Temp43 + "','" + Batt_Temp44 + "','" + Batt_Temp45 + "','" + Batt_Temp46 + "','" + String(air_quality_score) + "','" + mqtt_status1 + "','" + mqtt_status2 + "','" + mqtt_status3 + "','" + mqtt_status4 + "','" + (t - 3) + "','" + h + "','" + (p / 100) + "','" + SIV + "','" + cog + "','" + sog + "')";

}`

### Expected behavior

- Correct color parsing
- No effect of the format to the compiling process
- Autoformat has no effect on compiling process

### Arduino IDE version

2.0.1

### Operating system

macOS

### Operating system version

12

### Additional context

Sidenotes : 
- Special characters like ö ä ü are changed after save on the cloud and reload. Example: Für goes fÃ¼r
- I confessed i did not check with the latest nightly build because the release 2.0.1 was 6 days ago

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-02:
Hi @CptHolzschnauz. Thanks for your report.

>Wrong color parsing after col +/- 500

This is an intentional limit. The line length at which this occurs was recently reduced:

https://github.com/arduino/arduino-ide/pull/1450

It is possible to configure this via the advanced settings. I'll provide instructions:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open in the Arduino IDE main panel. In the "**Search Settings**" field, type `editor.maxTokenizationLineLength`
1. Adjust the value of the "**Editor: Max Tokenization Line Length**" setting according to your preference.
1. Click the **X** icon on the "**Preferences**" tab.

Please note that it was reduced for a good reason: https://github.com/arduino/arduino-ide/issues/1343
However, it was reduced from 20000 to 500, so it might be there is a higher value that will provide you with an acceptable balance between performance and highlighting.

>If it's on the same "width", the code from this line on is wrong parsed:

Please provide a minimal sketch that demonstrates the problem.

>If i autoformat that sketch it won't compile anymore.

Please provide a complete minimal sketch that demonstrates the problem. The sketch you provided is not valid code, so it is expected that it won't compile.

> Special characters like ö ä ü are changed after save on the cloud and reload. Example: Für goes fÃ¼r

This bug is being tracked here: https://github.com/arduino/arduino-ide/issues/449

CptHolzschnauz, 2022-11-03:
> However, it was reduced from 20000 to 500, so it might be there is a higher value that will provide you with an acceptable balance between performance and highlighting.

I set it to 750 and i suggest to use this as a new standard value. I can feel NO difference in the perfomance.
>Please provide a minimal sketch that demonstrates the problem. 
This problem is solved with the solution described above
>Please provide a minimal sketch that demonstrates the problem.
I can't. It's happen only with my megasketch (245240 Bytes). After Autoformat, compiler says he can't find a included library. Strange is also that the compiler warnings are in german while the IDE is set to english.
While testing to find out whats happening i did cmd+t and cmd+z (no coding) and then it was not compiling anymore with IDE2 (library not found). I closed IDE2, opend IDE1, compiled fine, closed IDE1 and open IDE , look what happen:
The output is tokenized and the sketch compiles again with no problem:
 
[image]

Something is rotten in tokenization/autoformatting/feeding the compiler. Sorry that i can't provide a proper error description, that behaviour is too flaky to describe.

per1234, 2022-11-04:
>I can't. It's happen only with my megasketch (245240 Bytes). After Autoformat, compiler says he can't find a included library.

Since it is not actionable based on the limited information we have, I will dedicate this issue exclusively to your request that the default `editor.maxTokenizationLineLength` value be increased.

>Strange is also that the compiler warnings are in german while the IDE is set to english

Thanks for bringing this to our attention. I have submitted a dedicated report for it: https://github.com/arduino/arduino-ide/issues/1636

In the future, please, please submit a separate report for each distinct issue. These rambling issues are very difficult to manage and make it likely we will lose track of one of the things that were reported.

CptHolzschnauz, 2022-11-05:
After reinstallation the issues with the IDE was gone. 
Ergo: Loading a sketch with a line longer than `editor.maxTokenizationLineLength` can corrupt the IDE.
Proposal: editor.maxTokenizationLineLength = 750

per1234, 2022-11-05:
Hi @CptHolzschnauz.

> After reinstallation the issues with the IDE was gone.

I'm glad it is working now.

> Proposal: `editor.maxTokenizationLineLength` = 750

If you close the issue, it will not be possible for us to track your proposal. So I will reopen it for you.

CptHolzschnauz, 2022-11-05:
Addition to the autoformat issue:
The code
```

char *Name_T4_array[9] =
{
  "SENSOR 1",
  "SENSOR 2",
  "SENSOR 3",
  "SENSOR 4"
};

```
Leads to compiling errors after autoformat while the code
```

char *Name_T4_array[9] = // WITH a comment here
{
  "SENSOR 1",
  "SENSOR 2",
  "SENSOR 3",
  "SENSOR 4"
};
```
does not, but only reproducable in my sketch (library not found), copied to a small example its fine - so very exotic and negligable.

CptHolzschnauz, 2022-12-04:
FYI: The strange behaviour i observed returned also with the new installed IDE. Out of a sudden no compiling, can't find lib, the 1.8 IDE compiles fine the same sketch, reopen in IDE 2 eveything is fine. Can't say why, can't reproduce. Only hint - it's a large sketch of 212 kB.

roboter-basteln, 2022-12-30:
I am against increasing the limit. I would even propose to decrease it.
I teach 10 year old children to not produce such long lines for reasons of readability and maintainability.
If you have such a long line, you have Clean Code problems, too. 

Similar for a 212 kB sketch. In that case, your project is too large for a single file. 
You are probably not a beginner developer any more. 
Start using object oriented programming and split your large codebase into smaller, more manageable pieces.
Switch to an IDE which can handle multiple files better than Arduino IDE.

CptHolzschnauz, 2022-12-30:
Copy that, but then it would be wise that the editor is not allowing to write over that limit. In my case, this issue ended up with compiler problems and IDE failures.

Long line: I have to construct a large String for a MQTT message and then IMHO it's a good reason for one long line. If you do CR's its strange to look a syntactically wrong expression on the new line and i also faced problems in IDEs thats why i don't like CR's in the middle of a line - for clean code (But you're right, mine is far away from clean code from a academic perspective... I think the computers are fast enough these days to handle a lenght of 750.

Large sketch: My code is divided in several files, pictures and fonts etc. The compiler crunches all them into one bin file at the end and telling you the size of that bin, so you can't conclude that a large sketch is just one file. There you have a misunderstanding.

Switch IDE: I code Arduino stuff with Visual Studio Platform IO for inline debuging, IDE 1 and 2. IDE 1 is very familiar and with a dark theme also fits for long sessions. IDE 2 needs IMHO some more progress (Serial Monitor, Split window  and colaberation with the cloud) but for sure IDE 2 is the future and i will contribute at least with comments, money and issue postings.

Have a good party all of you!