arduino/arduino-ide #1620: My library doesn’t appear in Examples from Custom Library on Arduino 2.0.0 and Arduino 2.0.1

zeynepdicle, 2022-11-03:


per1234, 2022-11-03:
Hi @zeynepdicle. The reason your examples are not shown in the menu is because they are not valid sketches.

As stated in the Arduino Sketch Specification:

https://arduino.github.io/arduino-cli/latest/sketch-specification/#primary-sketch-file

> Every sketch must contain a `.ino` file with a file name matching the sketch root folder name.

The file and folder names of your example sketches differ in case:

```text
examples/
├── KizilotesiAlici/
│   └── KizilOtesiAlici.ino
└── KizilotesiVerici/
    └── KizilOtesiVerici.ino
```

Note that the "`o`" character is lowercase in the folder names, but uppercase in the file names ("`O`").

Because Windows has a case insensitive file system, these two filenames are considered equivalent. This is why the sketch is considered valid in Arduino IDE 1.x. But we must give consideration to users of other operating systems. Linux is a case sensitive file system. Linux users will find that your examples are invalid even when using Arduino IDE 1.x, so the fact that Arduino IDE 2.x gives consistent treatment to sketches is beneficial.

closed 2022-11-03 by per1234

zeynepdicle, 2022-11-03:
Thank you @per1234 I could fixed