arduino/arduino-ide #1630: Enable All Warnings (-Wall -Wextra) by default

nuket, 2022-11-03:


brianrho, 2024-01-01:
This has been open for nearly 9 years now, and I raised a new issue #2316 which just got closed as a duplicate. Is there a reason it hasn't been fixed yet?

It seems like a fairly simple change to make, with no downsides and clear benefit for the platform.

https://github.com/arduino/arduino-ide/blob/73b6dc4774297e53f7ea0affdbc3f7e963b8e980/arduino-ide-extension/src/browser/arduino-preferences.ts#L113

----

To be clear, the request is to change this setting to "Default" generally, whether for cores, sketches, etc. Whether people choose to ignore the warnings or manually bump the level up to "-Wall"/"-Werror" should be entirely up to them, but they should have a chance to **see** these basic warnings first and potentially fix their mistakes. 

The default set of GCC warnings should suffice for most usecases (which is why they are default, in the first place), an okay balance for the platform. I don't see any reason why warnings would be any more scary than errors, which users already see and have to fix, so this shouldn't be an argument.

At worst, a user sees some warnings from some library (which compiles just fine) and if they're at all worried, they search or raise a question on some Q/A site where people explain whether the warning is harmless or not -- and/or better yet, they'll raise an issue against the library to find out what's going on, which should in time prompt authors to fix these things or find other ways to do stuff that don't provoke warnings.

This is certainly better than the silent alternative where you spend ages trying to figure out why your device is crashing, when a simple warning would've told you or provided hints immediately.

cousteaulecommandant, 2024-01-02:
I think this was never fixed due to lack of consensus.  Some views were that defaulting to `-Wall` shouldn't be done because it would trigger library warnings.  Others suggested that maybe libraries should be compiled without warnings and sketches with them, but that implies changing the way the Arduino toolchain works, and that's a lot of work so this caused loss of interest in the idea.  My opinion was that libraries shouldn't trigger warnings if they are properly written (they're supposed to be quality code that will stay around forever, not some quick sketch written by an inexperienced user; throwing warnings can be seen as not passing a minimum quality requirement).  And finally some think that any warning at all will put off novice users, and it's better to just let them figure out the mistakes the hard way.

Overall, I think that at the very least Arduino shouldn't purposefully disable "natural" warnings, and if a user triggers such a warning it probably means they did something catastrophic and should know about it - after all, Arduino is meant for learning.
(I'd personally push for -Wall to be the default for the same reason - these warnings can be didactic and help learn programming - but if it helps getting this issue resolved, I'd be OK with Default being the default as a compromise.)

brianrho, 2024-01-08:
Yeah, there's no consensus on `-Wall`, I don't see any reason to impose it on everyone, if even professional environments often don't use it.

The default set is already a good default, whether for sketches, libraries, etc., this doesn't need to get terribly complex. I hope it can get resolved soon and doesn't hang around for another decade.