arduino/arduino-ide #1642: (Feature Request) Arduino on Windows ARM

justrach, 2022-11-07:
Hi all, is there any Arduino support for Windows on ARM64 as of yet?

Or any way to compile a said binary for Arduino for ARM.

Thanks

Lim1029, 2022-12-10:
it's kinda sad that arduino IDE has arm64 version for Linux but not Windows

Rarisma, 2023-08-24:
Hey I know this is kind of old but the IDE does work on windows on arm, as windows 11 can emulate x64 and x86 apps; not very well mind you but it does work.

I'm using a Surface Pro X (win 11) and I can compile my code for my ESP32 board but it's incredibly slow, taking several minutes for a compile, I can't imaging compiling for ARM would be very hard but I assume someone has already tried that. A native ARM build would massively improve arduino development on these devices.

mikednonotthatmiked, 2023-09-06:
> Hey I know this is kind of old but the IDE does work on windows on arm, as windows 11 can emulate x64 and x86 apps; not very well mind you but it does work.
>
> I'm using a Surface Pro X (win 11) and I [...]

While I agree that it does work on ARM64, running in x64 mode on ARM is a tremendous performance and battery life penalty versus native code. Linux and MacOS offer ARM native versions - I feel like the team could provide Windows ARM versions with minimal added effort.

aufdenkampe, 2024-09-04:
+1 for Windows ARM64 builds. An increasing number of Windows laptops are using ARM64.