arduino/arduino-ide #1651: Support installing libraries from folders via "Add .ZIP Library"

per1234, 2022-11-08:
### Describe the request

Add support for installing libraries from folders via "**Add .ZIP Library**".

🙂 The IDE will ensure libraries in any format are installed correctly.

### Describe the current behavior

Three methods are available for installing libraries:

- [Library Manager](https://docs.arduino.cc/software/ide-v2/tutorials/ide-v2-installing-a-library)
- [**Sketch > Include Library > Add .ZIP Library...**](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#importing-a-zip-library)
- [manually](https://docs.arduino.cc/software/ide-v1/tutorials/installing-libraries#manual-installation)

Although Library Manager is the best option for installation of release versions of the thousands of libraries that are available for installation via that method. The alternative installation methods are required in some cases:

- Beta testing non-release development revisions of the library
- Installing libraries not available from Library Manager

Libraries can only be installed from a ZIP file via "**Add .ZIP Library**" if the ZIP archive has the correct structure of the library located in the root folder. Attempts to install from a ZIP file that has a library in a subfolder results in an error:

> **Error: Request installZip failed with message: 13 INTERNAL: Library install failed: library not valid**

Despite the "ZIP" in the name, the Arduino IDE 1.x "**Add .ZIP Library**" feature has an undocumented capability to install libraries from a folder in addition to installation from a ZIP file. Because the average Arduino user has great difficulty with manual installation, this is the best installation method in cases where the ZIP file does not have the correct structure.

🙁 The "**Add .ZIP Library**" feature of Arduino IDE 2.x does not allow installation from folders.

#### To reproduce

1. Download the popular GRBL CNC firmware library from the following link:
   https://github.com/gnea/grbl/archive/refs/heads/master.zip
1. Select **Sketch > Include Library > Add .ZIP Library...** from the Arduino IDE 2.x menus.
1. Select the downloaded `grbl-master.zip` file in the "**Select a zip file containing the library you'd like to add**" dialog.
1. Click the <kbd>**Open**</kbd> button.
   🤷 An error notification appears:
   > **Error: Request installZip failed with message: 13 INTERNAL: Library install failed: library not valid**

   **ⓘ** This is expected because the GRBL library is in [the `grbl` subfolder of the repository](https://github.com/gnea/grbl/tree/bfb67f0c7963fe3ce4aaf8a97f9009ea5a8db36e); not in the root as would be required to produce a valid ZIP archive structure.
1. Unzip the downloaded `grbl-master.zip` file.
1. Select **Sketch > Include Library > Add .ZIP Library...** from the Arduino IDE 2.x menus.
1. Select the `grbl` subfolder from the extracted folder in the right panel of the "**Select a zip file containing the library you'd like to add**" dialog.
1. Click the <kbd>**Open**</kbd> button.

🙁 The `grbl` subfolder opens in the dialog instead of the expected behavior of installing the library from the folder. There is no reasonable way to install this library via the Arduino IDE.

Note that, aside from steps 2-4 above which were added to demonstrate that the library can't be installed from the ZIP file, the steps above are the officially documented installation procedure:

https://github.com/gnea/grbl/wiki/Compiling-Grbl#via-the-arduino-ide-all-platforms-recommended-for-all-users

### Arduino IDE version

2.0.1

### Operating system

- Windows
- Linux

### Operating system version

- Windows 10
- Ubuntu 20.04

### Additional context

The same repository structure that makes the library impossible to install via "**Add .ZIP Library**" also makes it impossible to submit to Library Manager:

https://github.com/arduino/library-registry/blob/main/FAQ.md#what-are-the-requirements-for-a-library-to-be-added-to-library-manager

---

Another example of extremely popular libraries hosted in a repository that does not have the structure to produce a ZIP file usable for installation:

https://github.com/jrowberg/i2cdevlib/tree/master/Arduino

The library author is aware of the situation, but has not been able to manage the necessary restructuring (which requires splitting the repository into many separate repositories in this case): https://github.com/jrowberg/i2cdevlib/issues/311

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-08:
I created this issue to provide a public record of the project management decision made on this subject, which was previously not publicly visible.

During the associated discussion, @kittaakos explained that it is not possible to emulate the Arduino IDE 1.x behavior from a technical standpoint:

>We cannot support both file and folder selectors on Windows and Linux. This is an electron limitation. This is the way how the native filer selector dialogs work.

This means that providing a capability to install libraries from folders would require the addition of a dedicated menu item. It was determined the harm resulting from the increase in UI complexity outweighed the benefit of the feature for the following reasons:

- Most library repositories have been converted to the correct structure
- The Arduino IDE 1.x feature is undocumented and not easily discoverable by users so it is likely few use it.
  From my observations, the primary usage was in the instructions provided by community members assisting users in the installation of these problematic libraries. Those community members are perfectly capable of providing manual installation instructions, even if it is less convenient.

closed 2022-11-08 by per1234

Cabinfourus, 2023-02-25:
So what are we to do when we want to add outside zip folders??????

per1234, 2023-02-25:
If anyone needs assistance installing libraries, please post on the Arduino Forum:

https://forum.arduino.cc/c/using-arduino/installation-troubleshooting/18

I'm certain we will be able to help you over there.