arduino/arduino-ide #1653: Baudrate 74880

Zaby1990, 2022-11-08:
### Describe the request

I'm working with a ESP8266 right now. When I reset the microcontroller it sends something over the Serial Port. 

This [source](https://arduino.stackexchange.com/questions/61770/esp-01-not-waking-correctly-up-after-deep-sleep) says, that the Baud rate which the ESP8266 unses in this state is 74880. 

I would like to choose this baudrate.

### Describe the current behavior

The baudrate is not available in the dropdown menue in 2.0.1.
[image]

### Arduino IDE version

2.0.1

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-08:
Hi @Zaby1990. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/serial-monitor/issues/20.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-11-08 by per1234

Zaby1990, 2022-11-09:
Hi,

 

of course, it’s fine this way, thank you. Unfortunally I only searched for the missing Buadrate in the issues, so I missed the other issue.

 

I see we have another report about this at arduino/serial-monitor#20 <https://github.com/arduino/serial-monitor/issues/20> .