arduino/arduino-ide #1663: Include Windows device drives in IDE 2.0.x as in earlier versions

thorathome, 2022-11-10:
### Describe the request

New Dell Win 11 Home installation did not get drivers from my recent IDE 2.0.1 installation. 
Simply installing IDE 1.8.19 gave me the drivers for v2.0.1. Easy.  

However, this would be a showstopper for a newbie, suggest you include drivers in the Windows installation.

Thanks. I've moved to IDE 2.0.1, like it and use it. 

### Describe the current behavior

No device drivers mean Arduino devices are not seen by the IDE. 

To make it slightly worse, my Win 11 Home install shows a COM port in the IDE->Tools->Port menu that is used by Windows but is not available to the IDE, making this confusing for me and difficult for a newbie trying to connect a device. 

By installing IDE 1.8.19 and checking its load drivers option, I was able to use IDE 2.0.1. 
With IDE 2.0.1, I still see the (unavailable) Windows COM port, and can also see COM ports used by connected Arduino devices.  

Thanks. Nice work.  

### Arduino IDE version

Version: 2.0.1 Date: 2022-10-27T13:26:40.924Z CLI Version: 0.28.0 [06fb1909] 

### Operating system

Windows

### Operating system version

Windos 11 Home 22H2

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-10:
Hi @thorathome. Thanks for your suggestion.

At the time the drivers were added to Arduino IDE 1.x, that approach made sense because there was only the AVR boards, and the AVR boards support was bundled with the Arduino IDE installation package.

The boards platform ecosystem has expanded over the years to several official platforms and hundreds of 3rd party platforms. Arduino created [**Boards Manager**](https://docs.arduino.cc/learn/starting-guide/cores) as an easy way for the user to manage their boards platforms. Since the user might now install any arbitrary array of platforms, it no longer made sense to attempt to install and manage every possible driver for every board via Arduino IDE distributions. The far better system of bundling the drivers with the boards platform and having Boards manager install them was implemented. The system was also opened up to allow 3rd party boards platforms to install drivers.

So the drivers included in the Arduino IDE 1.x installation package were superfluous and there was no point in migrating this obsolete system over to Arduino IDE 2.x.

The relevant drivers are now installed along with the Arduino AVR Boards platform:

https://github.com/arduino/ArduinoCore-avr/tree/master/drivers

The Arduino AVR Boards platform is automatically installed the first time you run Arduino IDE, so from the user's perspective there really isn't a significant change in the fact that installing the IDE also results in those drivers being installed.

As you can now see, there is no need for your proposal of emulating the Arduino IDE 1.x drivers system. The question that remains is why the new system did not work for your board.

Which exact USB to serial adapter chip is used by your board?

This is a black chip near the USB socket.

The chip will usually be identified by writing on the top. This might say something like "**Atmel MEGA16U2**" or "**WCH CH340G**" or "**SILABS CP2102**" or "**FTDI FT232RL**".

Examples:

## [Microchip ATmega16U2](https://www.microchip.com/en-us/product/ATMEGA16U2)

[image]

## [WCH CH340](http://www.wch-ic.com/products/CH340.html)

[image]

## [FTDI FT232R](https://ftdichip.com/products/ft232rl/)

[image]

thorathome, 2022-11-12:
Thank you for the quick and detailed response. Much appreciated.
I have just moved to IDE 2.0.x and appreciate your effort. Quite an
upgrade.

I use Arduino with Adafruit ESP32 Feathers and generic ESP8266 devices.
After installing your IDE 2.0.1 on a new Win 11 Home installation, the IDE
did not detect the ESP32s. I then installed the IDE 1.x with its included
driver package, found that version detected the ESP32s without problems.
When I returned to the IDE 2.0.1, it detected the ESP32.

Is this what you needed?

per1234, 2022-11-15:
I guess your boards must use one of the Silicon Labs CP210x USB chips. The driver for those chips is included in the Arduino IDE 1.x installation:

https://github.com/arduino/Arduino/tree/master/build/windows/dist/drivers

When the drivers were added to "Arduino AVR Boards", those drivers were omitted because none of Arduino's boards use these chips:

https://github.com/arduino/ArduinoCore-avr/pull/442#issue-1083393575

> I removed CP210x drivers since we do not have anymore boards that require them

The new way is for the maintainers of platforms for boards that require a driver to distribute the driver as part of their platform, along with an installation script:

https://arduino.github.io/arduino-cli/latest/platform-specification/#post-install-script

Here is an example of one of those scripts:

https://github.com/arduino/ArduinoCore-avr/blob/master/post_install.bat

You can submit a request for addition of the drivers to the Espressif boards platform repositories.

closed 2022-11-15 by per1234

thorathome, 2022-11-15:
> arduino/ArduinoCore-avr#442 (comment)

Thanks for hearing me out. Good luck with the project.

I'll report bugs and issues as I see them.

Be well.