arduino/arduino-ide #1664: IDE 2.0 Please show entire OTA Port identifier

thorathome, 2022-11-10:
### Describe the request

IDE 2.0.1 Tools->Port menu shows the current connected Network (OTA) port by IP address only. 
When selecting from multiple Network ports, the device name including the IP address is visible.

It Would Be Nice if the Tools->Port menu also showed the connected device name with the IP address. 

As I have dozens of Arduino devices on my network, the device name is helpful to see what Network Port I am currently connected to. 

Thanks. Nice work. 

### Describe the current behavior

IDE 2.0.1 Tools->Port menu shows the current connected Network (OTA) port by IP address only. 
When selecting from multiple Network ports, the device name including the IP address is visible.

### Arduino IDE version

Version: 2.0.1 Date: 2022-10-27T13:26:40.924Z CLI Version: 0.28.0 [06fb1909]

### Operating system

Windows

### Operating system version

Windows 11 Home 22Hs

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-11:
Hi @thorathome. Thanks for your suggestion.

> It Would Be Nice if the Tools->Port menu also showed the connected device name with the IP address.

Let me make sure I understand your request. Are you referring to the part in the **Tools** menu I have circled in this screenshot of Arduino IDE 2.x?

[image]

I see that in Arduino IDE 1.x, the hostname (`esp32-b4e62dbf693d`) in this example is shown there in addition to the IP address:

[image]

thorathome, 2022-11-12:
THANK YOU for the detailed and clear response. I am impressed.

Yes, this is exactly what I refer to. With more than a few devices in the
Ports list, the longer presentation you have circled would be most helpful.

I just switched over to IDE 2.0.x and appreciated the redo and all your
efforts.
I work these days with the Adafruit ESP32 Feather series and generic
ESP8266s.
Thanks.