arduino/arduino-ide #1671: Compilation error: No connection established

aaronw2, 2022-11-13:
### Describe the problem

When attempting to build I am suddenly getting the following error without any attempt to compile:

Compilation error: No connection established

It does not matter whether I select to just compile/verify or upload.

The ESP32 S3 dev board is connected to /dev/ttyUSB8 and was working fine, then everything stopped. I see the device when I plug it in. Furthermore, minicom connects just fine and displays the output.
This is not a permission problem. The user is associated with the dialout group and I use serial ports extensively (for non-Arduino purposes). It also was working and suddenly stopped. This has happened to me on several occasions. Usually I am forced to restart the IDE to fix the issue.

Also, when building/verifying there should be no connection dependency.

Selecting a different serial port then selecting the correct one does not fix the problem.
Unplugging and reconnecting the ESP32 dev board (UART interface) does not fix the problem.
Only restarting the IDE fixes the problem.

### To reproduce

Work with the IDE for a while and upload multiple images. The problem is intermittent.

### Expected behavior

I expect it to not require a connection when building/verifying. Furthermore, I expect it to be able to connect to the serial port specified.

### Arduino IDE version

Version: 2.0.2-nighjtly-20221112

### Operating system

Linux

### Operating system version

OpenSUSE 15.4

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-14:
Hi @aaronw2. Thanks for taking the time to submit an issue.

I see we have another report about this at https://github.com/arduino/arduino-cli/issues/1970.

---

A fix for the bug has already been prepared. It would be very helpful if you would give the tester build of Arduino IDE with that fix a try and then let us know whether it solves the problem for you.

I'll provide the instructions for getting the tester build of Arduino IDE that has the proposed fix:

1. Sign in to your [**GitHub**](https://github.com/) account.
   (GitHub only allows downloads of the tester builds when you are signed in.)
1. Open this page in your browser:
   <https://github.com/arduino/arduino-ide/actions/runs/3444944271#artifacts>
1. Click the "**Linux_X86-64_app_image**" link under the "**Artifacts**" section of the page.
1. Wait for the download to finish.
1. Extract the downloaded file as usual.
1. Set the permissions on the `.AppImage` file to executable.
1. Start the tester build of Arduino IDE you downloaded.
1. Perform whatever operations normally produce the error when using the regular version of Arduino IDE.

After doing that, please comment here to let me know the following:

- Were you able to reproduce the fault using the regular version of Arduino IDE?
- Were you able to reproduce the fault using the tester version of Arduino IDE?

Please let me know if you have any questions or problems while following those instructions.

closed 2022-11-14 by per1234

collet1934, 2023-05-20:
I have the same problem.
I use IDE 2.1 and  W11 famille 21H2
My MKR zero is auto detected, the port is correctly detected and  the tools>>board info  is correct.
I  upload the codes without problem for a while ( # more than 1 h)
Then without notice I get the '' compil error: no connect...''.
The mkr zero and the comPort are correctly  detected, but I cannot compile and upload !!!
[image]
may I do something to help ?