arduino/arduino-ide #1674: Problem with Auto Format in Arduino IDE 2.0.1

Pexje, 2022-11-14:
### Describe the problem

When using Auto Format in Arduino IDE 2.0.1 some code in beginning of sketch goes wrong:

[image]

This gives Compilation error.

### To reproduce

Please fix this bug.

### Expected behavior

This is good working code before using Auto Format:

[image]

### Arduino IDE version

2.0.1

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-15:
Thanks for your report @Pexje.

>This gives Compilation error.

Although I am able to reproduce the formatting change you describe:

before

```cpp
#define DEBUG_SERIAL if (DEBUG) Serial
```

after

```cpp
#define DEBUG_SERIAL \
  if (DEBUG) Serial
```

I am not able to reproduce the compilation error you claim. I will need you to provide enough information for me to reproduce it:

- A complete minimal sketch that compiles before formatting and does not compile after formatting.
- Does the problem occur if you select **Tools > Board > Arduino AVR Boards > Arduino Uno** from the Arduino IDE menus before compiling the formatted sketch? If not, which board do you have selected when compilation does fail?
- The full an exact text of the compiler error message shown in the "**Output**" panel at the bottom of the Arduino IDE window.

## Investigation

I looked for a setting in [the **ClangFormat** options](https://releases.llvm.org/14.0.0/tools/clang/docs/ClangFormatStyleOptions.html) (ClangFormat is the tool used by the Arduino IDE "**Auto Format**" feature) that would cause it to not add this line break. I did not find any.

It is not clear whether the addition of this break is intentional behavior of the formatter or a bug. I found this similar bug report: https://github.com/llvm/llvm-project/issues/53180. I also found this related bug report https://github.com/llvm/llvm-project/issues/49164, which has been fixed, but that fix did not fix the problem you report here.

## Workaround

You can selectively disable formatting by adding special comments to your sketch code:

https://releases.llvm.org/14.0.0/tools/clang/docs/ClangFormatStyleOptions.html#disabling-formatting-on-a-piece-of-code

So if you don't like the line break, you can do this:

```cpp
// clang-format off
#define DEBUG_SERIAL if (DEBUG) Serial
// clang-format on
```

per1234, 2023-09-26:
Since the requested code demonstrating a functional change to the code was never provided, and since the formatting style is not objectively bad from an aesthetic standpoint, I will close this as declined/invalid.

If someone can provide a sketch that demonstrates the changed formatting causing a functional change to the code then we will investigate further.

closed 2023-09-26 by per1234