arduino/arduino-ide #1678: .s file do not get copied into build directory for project with ULP code

2dom, 2022-11-15:
### Describe the problem

When trying to compile ULP code, the .s file is not copied from the source directory into the build directory. Hence the compile fails with 'undefined reference to' errors. The linking succeeds if the file is copied into the build directory by hand. 

### To reproduce

Compile any ULP example project  

### Expected behavior

Compile succeeds

### Arduino IDE version

2.01

### Operating system

macOS

### Operating system version

Ventura 13.0.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://github.com/arduino/arduino-ide#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-16:
Hi @2dom. Thanks for your report. Assembly files must use the `.S` file extension (with an upper case "S"). The results you reported are normal and expected when you use the incorrect file extension.

closed 2022-11-16 by per1234

2dom, 2022-11-16:
No...ULPtool specifically states that the assembler files need to be .s (lower case) and the build file looks for those files in the build directory.

I'm not sure if this was a work around by ULPtool, however, if I rename it to .S the toolchain tries to compile a different way and fails.