arduino/arduino-ide #1691: Code would not compile/link without errors, copy of code compiled/linked without errors

meakerb, 2022-11-17:
### Describe the problem

Using IDE 2.0.0 (or 2.0.1?) on a Windows 11 machine. Had a large program that I broke into multiple files, with corresponding .h files, extern declarations, etx. Code would compile but fail during linking, giving 'multiple definition of' errors (error messages pasted below). Machine was powered-down/rebooted without changing the problem.
Today I copied the sketch folder and renamed it (along with the .ino file) so that I could experiment with changing the code around to see what would fix the error. When I started the IDE, I was notified of a new version of the IDE (2.0.2), so I did the update. When I opened the new copy of my code, it compiled and linked without error. But when I opened the original (identical) version, it still gives me errors.

Using board 'mega' from platform in folder: C:\Users\meakerb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Using core 'arduino' from platform in folder: C:\Users\meakerb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6
Detecting libraries used...
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\sketch\\Shirley_v2p0.ino.cpp" -o nul
Alternatives for LiquidCrystal_I2C.h: [LiquidCrystal I2C@1.1.2]
ResolveLibrary(LiquidCrystal_I2C.h)
  -> candidates: [LiquidCrystal I2C@1.1.2]
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LiquidCrystal_I2C" "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\sketch\\Shirley_v2p0.ino.cpp" -o nul
Alternatives for Wire.h: [Wire@1.0]
ResolveLibrary(Wire.h)
  -> candidates: [Wire@1.0]
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LiquidCrystal_I2C" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\Wire\\src" "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\sketch\\Shirley_v2p0.ino.cpp" -o nul
Alternatives for LibPrintf.h: [LibPrintf@1.2.13]
ResolveLibrary(LibPrintf.h)
  -> candidates: [LibPrintf@1.2.13]
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LiquidCrystal_I2C" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\Wire\\src" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LibPrintf\\src" "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\sketch\\Shirley_v2p0.ino.cpp" -o nul
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\button.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\initMenu.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\initScorecard.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\rotate.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\updateLCD.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\utils.cpp
Using cached library dependencies for file: c:\Users\meakerb\Documents\Arduino\libraries\LiquidCrystal_I2C\LiquidCrystal_I2C.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src\Wire.cpp
Using cached library dependencies for file: C:\Users\meakerb\AppData\Local\Arduino15\packages\arduino\hardware\avr\1.8.6\libraries\Wire\src\utility\twi.c
Using cached library dependencies for file: c:\Users\meakerb\Documents\Arduino\libraries\LibPrintf\src\LibPrintf.cpp
Generating function prototypes...
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -w -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -flto -w -x c++ -E -CC -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LiquidCrystal_I2C" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\Wire\\src" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LibPrintf\\src" "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\sketch\\Shirley_v2p0.ino.cpp" -o "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\preproc\\ctags_target_for_gcc_minus_e.cpp"
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\builtin\\tools\\ctags\\5.8-arduino11/ctags" -u --language-force=c++ -f - --c++-kinds=svpf --fields=KSTtzns --line-directives "C:\\Users\\meakerb\\AppData\\Local\\Temp\\arduino-sketch-65B68380ED218314C243A40E61648DDB\\preproc\\ctags_target_for_gcc_minus_e.cpp"
Compiling sketch...
"C:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\tools\\avr-gcc\\7.3.0-atmel3.6.1-arduino7/bin/avr-g++" -c -g -Os -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto -mmcu=atmega2560 -DF_CPU=16000000L -DARDUINO=10607 -DARDUINO_AVR_MEGA2560 -DARDUINO_ARCH_AVR "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\cores\\arduino" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\variants\\mega" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libraries\\LiquidCrystal_I2C" "-IC:\\Users\\meakerb\\AppData\\Local\\Arduino15\\packages\\arduino\\hardware\\avr\\1.8.6\\libraries\\Wire\\src" "-Ic:\\Users\\meakerb\\Documents\\Arduino\\libr [... cut ...]
Not sure how to reproduce this error, since a copy of my code compiles/links without error.

### Expected behavior

Code should compile/link consistently.

### Arduino IDE version

2.0.2 

### Operating system

Windows

### Operating system version

10.0.22621 Build 22621

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-18:
Hi @meakerb. Thanks for your report. Please provide a list of the filenames of all the files in your sketch.

meakerb, 2022-11-19:
The files are:
Shirley_v2p0.ino
button.cpp
button.h
defines.h
initScorecard.cpp
initScorecard.h
main.h
rotate.cpp
rotate,h
updateLCD.cpp
updateLCD.h
utils.cpp
utils.h

I'm not sure it's relevant, but in the original sketch (that doesn't link properly) I rearranged the file tabs in the IDE. For the copy (that does link properly) I did not rearrange any tabs. Thanks.

per1234, 2022-11-19:
Thanks for the information. You can see the problem from this part of the compiler errors:

> ```text
> (.text+0x0): multiple definition of `initMenu()'
> C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB\sketch\initMenu.cpp.o (symbol from plugin):(.text+0x0): first defined here
> ```
Notice that it is referring to a file named `initMenu.cpp`, yet you don't have any file of that name in your sketch. This is a bug in the Arduino CLI tool that orchestrates the compilation. It does not clean up the build folder correctly after you rename a non-`.ino` sketch file:

https://github.com/arduino/arduino-cli/issues/1240

The workaround is to manually delete the sketch build folder shown in the error message:

```text
C:\Users\meakerb\AppData\Local\Temp\arduino-sketch-65B68380ED218314C243A40E61648DDB
```

❗ If looking for it with your file browser or command line, note that the `C:\Users\meakerb\AppData` folder is hidden by default. On Windows "**File Explorer**", you can make it visible by opening the "**View**" menu, then checking the box next to "**☐ Hidden items**".

The build folder will be automatically regenerated when you compile your sketch again, so the error should no longer occur after you do that.

---

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of https://github.com/arduino/arduino-cli/issues/1240.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-11-19 by per1234