arduino/arduino-ide #1697: [Serial Plotter] Request for XY-plotter

johan1937, 2022-11-18:


per1234, 2023-03-15:
I believe this was closed by accident because I don't see that https://github.com/arduino/arduino-ide/commit/2aad0e3b163f27c51b15b58a6f35aab757a643fb provides the requested functionality.

kittaakos, 2023-03-16:
> I believe this was closed by accident because I don't see that [2aad0e3](https://github.com/arduino/arduino-ide/commit/2aad0e3b163f27c51b15b58a6f35aab757a643fb) provides the requested functionality.

Thank you! For the record, https://github.com/arduino/arduino-ide/pull/1927 should have closed #1679 instead of #1697.

Feel free to hide this comment if off topic. I wanted to leave some info on why it was closed accidentally.

commit 5ca1e222 (2023-03-07): feat: new UX for the boards/library manager widget

Closes #19
Closes #781
Closes #1591
Closes #1607
Closes #1697
Closes #1707
Closes #1924
commit e9a6bcb1 (2023-03-09): feat: new UX for the boards/library manager widgets

Closes #19
Closes #781
Closes #1591
Closes #1607
Closes #1697
Closes #1707
Closes #1924
Closes #1941
commit 14e3476a (2023-03-09): feat: new UX for the boards/library manager widgets

Closes #19
Closes #781
Closes #1591
Closes #1607
Closes #1697
Closes #1707
Closes #1924
Closes #1941
commit 1d2de775 (2023-03-09): feat: new UX for the boards/library manager widgets

Closes #19
Closes #781
Closes #1591
Closes #1607
Closes #1697
Closes #1707
Closes #1924
Closes #1941