arduino/arduino-ide #1698: No Install option for ESP8266 board

OslafEman, 2022-11-18:
### Describe the problem

I have tried to install the ESP8266 board several times without success on the Arduino IDE 2.01 and 2.02.  After entering http://arduino.esp8266.com/stable/package_esp8266com_index.json in Preferences -> Additional URLs, when I try to install the ESP8266 the option "Install" (the Install button) does not appear.  All other boards can be installed without problem.  I've tried on Windows 10 and 11.

### To reproduce

Windows 10 and Windows 11
Arduino IDE 2.0.1 and 2.0.2

### Expected behavior

The Install option should appear for de ESP8266 board

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

22H2

### Additional context

Non

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-19:
Hi @OslafEman. Thanks for your report. Please try this:

1. Select **Tools > Board > Boards Manager...** from the Arduino IDE menus to open the Boards Manager view if it is not already open.
1. Hover your mouse pointer over the right border of the Boards Manager view until the pointer changes to the horizontal resize double arrow.
1. Click and drag the mouse pointer to the right to make the Boards Manager view wider.

After doing that, do you now see the **INSTALL** button?

OslafEman, 2022-11-19:
Thank you very much for the help @per1234, I followed the advice and I was able to install the ESP8266 core without problem.  The install button was just hidden.

per1234, 2022-11-19:
Thanks for the update. 

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1591.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-11-19 by per1234

scottchiefbaker, 2023-02-15:
I just ran in to this same "cannot install" issue today, and resizing the window fixed the problem.