arduino/arduino-ide #1699: Unable to delete tab

twanek, 2022-11-19:
### Describe the problem

In the latest IDE (2.0.2) when I select a tab, open the ... menu in the right, select Delete, click Ok, nothing happens. 

### To reproduce

select a tab, open the ... menu in the right, select Delete, click Ok

### Expected behavior

to delete the selected tab

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

10,  latest

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-19:
Hi @twanek. There are two distinct types of sketch editor tabs in Arduino IDE:

### Primary sketch tab

- This tab has the same name as the sketch
- It is normally the rightmost tab on the tab bar

### Secondary sketch tabs

- Name different from the sketch name

---

Please tell me which type(s) of tab you have selected when you experience this bug.

twanek, 2022-11-19:
Hi!

I'm talking about these ones:
[image: image.png]
when I try to delete one, it doesn't works...
I also tried to re-start the IDE with admin rights, makes no difference.

per1234, 2022-11-19:
Hi @twanek. GitHub removes attachments from emails so I can't see the picture you shared.

Please follow this link to the GitHub website and make the attachment in a comment there:

https://github.com/arduino/arduino-ide/issues/1699

twanek, 2022-11-19:
please see here:
[image]

per1234, 2022-11-19:
Thanks for the clarification. I asked because there is a known bug with deleting the primary sketch tab: https://github.com/arduino/arduino-ide/issues/649

From your screenshot, I can see that you have a secondary sketch tab selected, so this is something different from https://github.com/arduino/arduino-ide/issues/649

Unfortunately I have not been able to reproduce the bug.

---

I would now like to determine whether the problem only occurs for you with this one specific sketch, or does it occur with any sketch.

Please try this and tell me the result:

1. Select **File > New Sketch** from the Arduino IDE menus.
1. Click the **●●●** icon on the right side of the editor toolbar.
1. Select "**New Tab**" from the menu.
1. In the "**Name for new file**" field, type `secondary`
1. Click the <kbd>**OK**</kbd> button.
1. Click the **●●●** icon on the right side of the editor toolbar.
1. Select "**Delete**" from the menu.
1. Click the <kbd>**OK**</kbd> button.

After you do that, was the **secondary.ino** tab removed from the sketch?

per1234, 2022-12-16:
Hi @twanek. We are still waiting for you to provide the information requested in the previous reply so that we can proceed with the investigation or evaluation of this.

twanek, 2022-12-16:
Hello!

I know, sorry, but I stopped using the new IDE. It is very very slow and I
found several other bugs, so I reverted back to the old one.

best regards,