arduino/arduino-ide #1708: Arduino IDE is stuck on loading

Zhek1ch, 2022-11-25:
### Describe the problem

Today when I tried to open Arduino, it opened, but didn't show anything. MacOS menubar shows arduino ide as a focused app, but nothing is happening, I can't create a new project, cant open the recent ones. I also tried opening a random project on my computer, but then only the white screen is shown, and nothing else. Here is the image showing it.
[image]
Reinstalling IDE, restarting MacOS, deleting all arduino config files didn't work either.
The logs show me this:
`2022-11-25 18:27:24 root ERROR Failed to start the backend application:
2022-11-25 18:27:24 root ERROR Error: getaddrinfo ENOTFOUND localhost
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:71:26)
2022-11-25 18:27:24 root ERROR Uncaught Exception: 
2022-11-25 18:27:24 root ERROR Error: getaddrinfo ENOTFOUND localhost
    at GetAddrInfoReqWrap.onlookup [as oncomplete] (node:dns:71:26)`
Found the same problem in #1527, but it doesn't contain the fix for the problem.

### To reproduce

I dont know how it happened, the application just randomly stopped working.

### Expected behavior

I was expecting the application to load as normal.

### Arduino IDE version

2.0.0 - 2.0.3

### Operating system

macOS

### Operating system version

Big Sur 11.7.1

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-11-26:
Hi @Zhek1ch. Thanks for your report. Just in case they might provide some additional relevant information, please provide the complete logs.

Zhek1ch, 2022-11-26:
[2022-11-26_log.log](https://github.com/arduino/arduino-ide/files/10096917/2022-11-26_log.log)
Here is the full log

per1234, 2022-11-26:
Thanks. I see you did a good job of extracting all the relevant lines to include in your original report, but I had to check to be sure.

Did you modify the `/private/etc/hosts` (AKA `/etc/hosts`) file recently?

Zhek1ch, 2022-11-26:
No, the only thing I did was resetting the nvram, and after this some applications started glitching, and sadly arduino is one of them

kittaakos, 2022-11-29:
@Zhek1ch, please help us to narrow down the problem. IDE2 builds on the [Theia IDE](https://theia-ide.org/). From the logs, it seems the problem is with pure Theia. Could you please get the latest Theia Blueprint version from https://theia-ide.org/docs/blueprint_download, start it and let us know if the app loads or not?

It's a guess, but handling IPv6 addresses in Theia might have flaws. Thank you!

Zhek1ch, 2022-11-29:
No, Theia IDE doesn't load either
[image]
The application is active, but I can't do anything.

Zhek1ch, 2022-11-30:
Binding localhost to 127.0.0.1 fixed the problem
[Tutorial](https://setapp.com/how-to/edit-mac-hosts-file)

closed 2022-11-30 by Zhek1ch