arduino/arduino-ide #1711: Linux DEB and RPM support

ArduFish123, 2022-11-26:
### Describe the request

It would be great to have DEB and RPM installer packages for Arduino IDE.

### Describe the current behavior

There is no such package.

### Arduino IDE version

Any version

### Operating system

Linux

### Operating system version

Ubuntu 22.04 LTS

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

hlovdal, 2022-12-25:
With the appropriate rpm spec file added the [COPR build system](https://copr.fedorainfracloud.org/) could be utilised to do the actual building.

ArduFish123, 2023-06-30:
Duplicate issue. Order issue created at the link https://github.com/arduino/arduino-ide/issues/641

closed 2023-06-30 by ArduFish123