arduino/arduino-ide #1712: Display code errors in red in real time

JulienLecoq, 2022-11-27:
### Describe the request

I would like to have code errors appears in real time on screen (before compilation as does other IDEs)

### Describe the current behavior

Code errors are only printed inside the Output section of the IDE on Verify/Upload buttons click.

### Arduino IDE version

2.0.2

### Operating system

Linux

### Operating system version

I'm using Garuda Linux, according to lsb_release -a : "Talon"

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-11-27:
Hi @JulienLecoq. Thanks for your suggestion.

I'm happy to tell you Arduino IDE already has this capability. It can be enabled via the advanced settings:

1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Preferences: Open Settings (UI)**" command from the menu.
1. A "**Preferences**" tab will now open in the Arduino IDE main panel. In the "**Search Settings**" field, type `arduino.language.realTimeDiagnostics`
1. Check the checkbox under the "**Arduino › Language: Real Time Diagnostics**" setting.
1. Close the **Preferences** tab by clicking its **X** icon.

It is important to understand the reason why it was decided to have this setting off by default. [Library discovery](https://arduino.github.io/arduino-cli/latest/sketch-build-process/#dependency-resolution) is a convenient feature of the Arduino build system. It allows you to use any library in your sketch simply by adding an `#include` directive for a header file. In order to accomplish that, the Arduino IDE must search through all the library locations.

Previously, Arduino IDE 2.x triggered the library discovery process on every change to the sketch code. In order to reduce the resource usage, the IDE has now been changed to use only the following events as library discovery triggers (https://github.com/arduino/arduino-ide/pull/1107):

- Opening a sketch.
- Changing the board selection.
- Compiling the sketch

For this reason, the language server features will not work for sketch code that uses a library during the interval between when its `#include` directive was added to the sketch and when the library discovery is next triggered.

The deferred library discovery technique described above results in spurious problem detection for uses of the library in the sketch code before a discovery process is triggered.

closed 2022-11-27 by per1234

JulienLecoq, 2022-11-28:
Oh okay, thanks for your very detailed response @per1234 !