arduino/arduino-ide #1723: Syntax coloring stops after long Strings

francwalter, 2022-11-30:
### Describe the problem

I have Arduino IDE 2.0.2 and I have some long concatenated Strings. At the end of the String I can see that the Syntax Coloring stops working, all is brown now:

[image]

Here is the end of that string:

[image]

### To reproduce

Create a very long string and see how the Syntax Coloring stops working after.

### Expected behavior

It should still color variables etc.

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-11-30:
> I have some long concatenated Strings

Can you share a snippet? Maybe related https://github.com/arduino/arduino-ide/issues/1614

francwalter, 2022-11-30:
You are perfectly right, this is a dupe :)
Sorry for it, didnt find it.
Could just resolve it with:
CTRL+SHIFT+P > Preferences: Open Settings (UI) > editor.maxTokenizationLineLength = 750 (UI / Workspace)
Closed the IDE and reopen and all colors back again :)
My line was 619 characters, by the way.
Thank!
frank

closed 2022-11-30 by francwalter

CptHolzschnauz, 2022-11-30:
750 is IMHO a fair value for perfomance vs. avoid this issue. In my case, it led to strange compiler behaviour vs. a minority have this long lines. Computers are fast nowadays.

kittaakos, 2022-12-01:
> In my case, it led to strange compiler behaviour

Syntax coloring and compilation are unrelated. Do you have a reproducible example, I am happy to look into it.

CptHolzschnauz, 2022-12-01:
No, not really reproducable (Just happend on one system). 
The behaviour was:
- Unformatted : Compiles
- Autoformatted : Error from the compiler
At the end, the ide went mad with formatted serial monitor and flaky behaviour:

[image]

After reinstall, it was fine again, so not worth your time to sort out.

Ergo: Root cause for the IDE madness was, IMHO, that too long line, that's why i suggest 750 
https://github.com/arduino/arduino-ide/issues/1614

PS: I'll try to reproduce later and post a video, i can't share the source.
PS2: Thank you for all your effort and contributions kittakoos

kittaakos, 2022-12-01:
> The behaviour was:
> 
> * Unformatted : Compiles
> * Autoformatted : Error from the compiler
>   At the end, the ide went mad with formatted serial monitor and flaky behaviour:

The formatter might have bugs and makes compiler errors (https://github.com/arduino/arduino-ide/issues/1674), but the formatter is not the same as the syntax coloring. I think we are talking about different things. Anyway, if you have a reproducible sketch you can share, please do it, and I check what's going on.

francwalter, 2022-12-01:
I have a different question, a small one, dont want to open an issue and I dont find a discussion site in this repo:
In the Workspace Editor, when I press Enter to make a new line and the cursor is first at the identation of my code, but jumps to the very beginning of the next line, after about 2 seconds. 
Can I disable this (maybe nice) feature?
Makes me more often use the arrow-keys 😏 
Thank

kittaakos, 2022-12-01:
> I have a different question, a small one, dont want to open an issue

No need to: https://github.com/arduino/arduino-ide/issues/437 ;)

> Can I disable this (maybe nice) feature?

This is very much likely the same issue as https://github.com/clangd/vscode-clangd/issues/357, and it is not directly done by IDE2 but by the [Arduino Language Server](https://github.com/arduino/arduino-language-server/). We need to look into how to intercept and tweak the [`textDocument/completion`](https://microsoft.github.io/language-server-protocol/specifications/lsp/3.17/specification/#textDocument_completion). But this is just a guess so far.