arduino/arduino-ide #1724: IDE SerMon fails to UGI draw initial text output

Defragster, 2022-12-01:
### Describe the problem

Sketch that Serial.prints in setup()  has output present, but not visible in GUI Tab 'Serial Monitor' on initial change to that tab.

### To reproduce

See https://forum.pjrc.com/threads/71588-Arduino-IDE2-Serial-Monitor-sometimes-does-not-work-continue-from-Upload-Thread?p=316565&viewfull=1#post316565

Sketch provided there and steps showing repro on WINDOWS IDE 2.0.2 with Teensy 4.1.

Sketch uploads and starts sending out some 20+ lines but displayed output is BLANK up to the 10th line - image in post above.

Happens the SAME with IDE SerMon or Teensy Serial with output to that tab.

Hitting mouse Scroll wheel, or other, forcing redraw and the content is accurately shown.

HINT Perhaps?: It as if the output was displayed and scrolled down - only the tab was not in focus. So the 'scrolled view' 'shows' the lines that would have been displayed had the window scrolled down - but the tab window is instead at the top.

### Expected behavior

All buffered output should be visibly displayed in SerMon tab.

Other variations of this have been reported on Teensy Forum - this is a new thread and custom repro sketch with steps.

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

windows 11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

Defragster, 2022-12-01:
See https://forum.arduino.cc/t/ide-2-0-2-serial-monitor-data-forgets-to-scroll-with-new-data/1060247  and https://forum.pjrc.com/threads/71588-Arduino-IDE2-Serial-Monitor-sometimes-does-not-work-continue-from-Upload-Thread?p=316574&viewfull=1#post316574

As posted above, AutoScroll was ON and that was not noted as problem. Focus above was failure to draw 'top lines', but those are OLD lines and new data below should have scrolled up and did not.

So seems this should have been the real focus of this issue. Will change the title.

PaulStoffregen, 2022-12-02:
Shared this video on the forum threads, might as all have it lined here too.

Problem not limited to Teensy, also happens with Arduino Leonardo.

https://www.youtube.com/watch?v=oHIKPysz9dI

Defragster, 2022-12-05:
Updated to IDE 2.0.3 and the Blank/Clipped issue seems gone as the initial 'window' of text is now visible. At least this is the case with Updated PJRC Teensy Serial Monitor code.

The IDE Serial Port Monitor code however once repeated the 'BLANK' window? Now is showing initial sketch output?

However, there is a problem with AutoScroll: It scrolls for initial output from setup() - but stops there as additional output arrives it never moves to the scrolled bottom newest text?

Code below prints an "A=" section then a "B=" section, it scrolls to "A=" then stops leaving the following "B=" portion offscreen.

This can also be seen in that sketch when USB 'Message' is sent to the Teensy, which is echoed back. That arriving output stays offscreen and doesn't autoscroll into view.

This behavior is changed from IDE 2.0.2:
`void setup() {
  pinMode( LED_BUILTIN, OUTPUT );
  digitalWrite( LED_BUILTIN, HIGH );
  Serial.begin(115200);
  while (!Serial);
  digitalWrite( LED_BUILTIN, LOW );
  Serial.println("\n" __FILE__ " " __DATE__ " " __TIME__);
  Serial.print( "run secs A=" );
  Serial.println( millis()/1000.0 );
  delay(1000);               // wait for a second
  Serial.println("\n" __FILE__ " " __DATE__ " " __TIME__);
  Serial.print( "run secs B=" );
  Serial.println( millis()/1000.0 );
  showSome( 3 );
}

void loop() {
  // put your main code here, to run repeatedly:
  if ( Serial.available() )
    showSome( 3 );
  digitalWrite(LED_BUILTIN, HIGH);   // turn the LED on (HIGH is the voltage level)
  delay(1000);               // wait for a second
  digitalWrite(LED_BUILTIN, LOW);    // turn the LED off by making the voltage LOW
  delay(500);               // wait for a second
}

void showSome( int cnt ) {
  static uint nn=1;
  int ch;
  while ( (ch=Serial.read()) > -1 ) {
    Serial.print( (char)ch );
  }
  int ii;
  for ( ii=0; ii<cnt; ii++ ) {
    Serial.print( "See Me: " );
    Serial.println( ii+nn );
  }
  nn+=ii;
}`

Defragster, 2022-12-05:
Generally no, except in some amusing pattern:

Upload sketch (above) T_4.1 with IDE SerMon active - minimal out presented shows:
`
C:\T_Drive\tCode\FORUM\serialIDE2test\serialIDE2test.ino Dec  5 2022 12:12:56
run secs A=1.06

C:\T_Drive\tCode\FORUM\serialIDE2test\serialIDE2test.ino Dec  5 2022 12:12:56
run secs B=2.06
See Me: 1
See Me: 2
See Me: 3`

Upload again output in view advances only to:
`C:\T_Drive\tCode\FORUM\serialIDE2test\serialIDE2test.ino Dec  5 2022 12:14:04
run secs A=1.07`

Go to Message and send 'Hello' - output advances to prior:
`C:\T_Drive\tCode\FORUM\serialIDE2test\serialIDE2test.ino Dec  5 2022 12:14:04
run secs B=2.07
See Me: 1
See Me: 2
See Me: 3`

Go to Message and send 'Goodbye' - output advances to prior:
`Hello
See Me: 4
See Me: 5
See Me: 6`

Mouse wheel Scroll down to see:
`Goodbye
See Me: 7
See Me: 8
See Me: 9`

Go to Message and send 'Arduino' - output does not advance.

Same here IDE 2.0.3 Win 11 with IDE and PJRC Teensy Ports.

PaulStoffregen, 2022-12-08:
Confirm, also seeing poor autoscroll behavior with IDE 2.0.3.  Could record another video, if that would help?

per1234, 2022-12-09:
Hi @PaulStoffregen. The defect that is the subject of this issue: "Serial Monitor does not Autoscroll on new content received when view is not in focus" is already quite clear to me and easy to reproduce. So no additional clarification is needed. It is only necessary to wait for someone to prepare the code changes to fix the defect.

As for what appears to possibly be a separate defect reported at https://github.com/arduino/arduino-ide/issues/1724#issuecomment-1338032989, I have not been able to understand it. If that is indeed a separate defect, please do submit a separate issue for it with a clear description of the problem and detailed instructions we can follow to reproduce it. We should be careful to keep each issue focused on a single subject because otherwise they become very difficult and time consuming for the team to manage.

Defragster, 2022-12-09:
That last @per1234  comment "keep each issue focused on a single subject " refers to this #1724? The title was changed 6 days ago - which changed what the issue covered?

This is okay as it seems a change from 2.0.2 to 2.0.3 has eliminated the problem related to this issue where the initially reported 'drawing on an out of focus Sermon display window' resulted in seeing a blank window when it was put in focus. Now the window text is updated and draws - but per new title and recent comments it just fails to automatically scroll to the latest arriving information.

Seems for my last post here I did see a blank screen once - but all will be well perhaps in the end if auto scrolling starts working correctly as it seems to be a variant of the same evolving behavior of this issue. Indeed, if not then a new issue makes sense with then current details to repro.

commit 656357dc (2023-03-03): fix: update monitor output after widget show

Closes #1724
PR #1928 (2023-03-07): fix: scroll to the bottom after the state update
### Motivation

An alternative version of #1889. This PR also fixes #1724.

Scroll to the bottom fix:

https://user-images.githubusercontent.com/1405703/222668221-46b68d3b-ce4b-48ea-ba67-35f6d22b8667.mp4

Update widget content and scroll position after the widget shows:

https://user-images.githubusercontent.com/1405703/222668321-274bd881-f707-498f-bf3b-7b77af29b7a4.mp4

### Change description

Scroll to the bottom of the monitor widget after the state update.

### Other information

Closes #1724
Closes #1736

### Reviewer checklist

* [ ] PR ad [... cut]
commit 24dc0bbc (2023-03-07): fix: update monitor output after widget show

Closes #1724
closed 2023-03-07 by kittaakos