arduino/arduino-ide #1733: Restore support of `breadcrumbs.enabled`, `window.titleBarStyle`, and `workbench.editor.highlightModifiedTabs` advanced settings

kittaakos, 2022-12-01:
### Describe the request

Originally from [here](https://github.com/arduino/arduino-ide/pull/1662#pullrequestreview-1197074739):

> Partially resolves:
> 
> * #1077
>   
>   * Everything reported in the issue is resolved _except_ the following:
>     
>     * `breadcrumbs.enabled` has no effect
>     * `window.titleBarStyle` has no effect
>     * `workbench.editor.highlightModifiedTabs` has no effect

Reply from [here](https://github.com/arduino/arduino-ide/pull/1662#issuecomment-1331942472):

> > * `breadcrumbs.enabled` has no effect
> 
> The breadcrumbs feature has been forcefully removed from IDE2.
> 
> > * `window.titleBarStyle` has no effect
> 
> This has also been removed long ago.
> 
> > * `workbench.editor.highlightModifiedTabs` has no effect
> 
> I do not know what's this.
> 
> I can create a follow-up with these items.

Closing with the following comments from [here](https://github.com/arduino/arduino-ide/pull/1662#issuecomment-1333680106):

> > > breadcrumbs.enabled has no effect
> > 
> > 
> > The breadcrumbs feature has been forcefully removed from IDE2.
> > > window.titleBarStyle has no effect
> > 
> > 
> > This has also been removed long ago.
> 
> If the settings are not expected to work, I think it is reasonable to explain that in #1077 and consider documenting the fact in the issue report to be sufficient. These settings are intended to be targeted exclusively to advanced users so we don't need to be so concerned about the user experience being polished.
> 
> I don't know whether any further action to improve on the current situation is possible.
> 
> > > workbench.editor.highlightModifiedTabs has no effect
> > 
> > 
> > I do not know what's this.
> 
> You can see the feature in effect in VS Code here (I am using the "**Light+ (default light)**" theme in the screenshot, but I also see the border in all the other themes I checked: "**Dark (Visual Studio)**", "**Light (Visual Studio)**", and "**Dark High Contrast**"):
> 
> [image]

----

### Describe the current behavior

See the request description 👆 

If restoring the lost features is trivial, this should be covered in this task. If it's not trivial or it causes performance issues, etc., then the reason for the missing features should be explained in #1077.

### Arduino IDE version

5695fd8

### Operating system

macOS, N/A

### Operating system version

12.5.1

### Additional context

Related to #1077

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

kittaakos, 2023-02-09:
> * `window.titleBarStyle` has no effect

This is currently not supported by Theia on macOS. See [here](https://github.com/eclipse-theia/theia/blob/master/packages/core/src/electron-browser/window/electron-window-preferences.ts). The preference is connected to another setting value: [`window.menuBarVisibility`](https://github.com/eclipse-theia/theia/blob/52d6e877371204045d9b047b39f3d54672387258/packages/core/src/browser/core-preferences.ts#L83). This preference is also [disabled](https://github.com/eclipse-theia/theia/blob/52d6e877371204045d9b047b39f3d54672387258/packages/core/src/browser/core-preferences.ts#L95) in Theia on macOS. Both preferences must be supported, or there is no way to show a custom title bar.

I put together a preview from the latest Theia sources, and somebody has to decide what IDE2 wants to do with the preference once they are available. Please look at how it works on resizing concerning overlay, menu, etc.:

https://user-images.githubusercontent.com/1405703/217823484-d0bb6435-b7f8-4d05-81bf-6d9beb1030b1.mp4

 - On macOS, the window control buttons must be moved to the left, and custom button styles are needed to provide a native look and feel.
 - Figure out if IDE2 wants to support menus on the title bar. If not, the styling issues 👆 still be fixed on Windows and Linux.

kittaakos, 2023-02-15:
> * `workbench.editor.highlightModifiedTabs` has no effect

Upstream: https://github.com/eclipse-theia/theia/issues/12185

kittaakos, 2023-09-19:
From https://github.com/arduino/arduino-ide/issues/2229#issue-1899608371:

> ### Describe the request
> the menu bar looks somewhat out of place if your theme's backgound differs from pure grey [image]
> 
> ### Describe the current behavior
> the menu bar is not effected by your theme
> 
> ### Arduino IDE version
> 2.2.1
> 
> ### Operating system
> Windows
> 
> ### Operating system version
> 10

per1234, 2025-04-20:
Since it is a significant and complex task on its own, which doesn't appear to be inherently linked to the task of enabling the `breadcrumbs.enabled` and `workbench.editor.highlightModifiedTabs` settings, I split the component of this issue regarding adding support for `window.titleBarStyle` to a dedicated issue: https://github.com/arduino/arduino-ide/issues/2716, where it can be more effectively and efficiently tracked.