arduino/arduino-ide #1734: Toggle Autoscroll: Visible state ON/OFF

CptHolzschnauz, 2022-12-04:
### Describe the request

With the IDE 1.8x autoscroll was controlled with a checkbox where the user can see the state (Checked/Unchecked)
I would like to suggest a GUI element to see if autocroll is on or off. Crossed symbol, different symbol, colorized..whatever.

### Describe the current behavior

You can't see in IDE 2 if autoscroll is turned On or Off. 

### Arduino IDE version

2.03 Nightly build

### Operating system

macOS

### Operating system version

12

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-04:
Hi @CptHolzschnauz. Thanks for your report. Please tell me which theme you have selected from the "**Theme**" menu in the **File > Preferences** dialog in Arduino IDE.

CptHolzschnauz, 2022-12-04:
Dark (Theia)

per1234, 2022-12-04:
OK, then this is a duplicate of https://github.com/arduino/arduino-ide/issues/1053.

That issue was closed because the project managers decided that we should focus the available resources on maintaining Arduino's own custom themes and that the "Theia" themes will be removed from the Arduino IDE installation as soon as doing so is feasible (there currently are some technical challenges that are blocking that): https://github.com/arduino/arduino-ide/issues/1283

The supported themes are:

- **Dark (Arduino)**
- **Light (Arduino)**
- **High Contrast (Theia)**

The first two indicate the state off the "Autoscroll" setting via the background color on the icon. 

The last indicates the state off the "Autoscroll" setting via the outline of the icon.

If you discover any problems while using a supported theme, please do let us know.

closed 2022-12-04 by per1234

CptHolzschnauz, 2022-12-04:
Got it. Thanks for your attandence and time. Solved.