arduino/arduino-ide #1735: "File > Preferences" disabled if dialog used soon after IDE startup

per1234, 2022-12-04:
### Describe the problem

If the "**Preferences**" dialog is opened soon after Arduino IDE is started, after closing the menu the **File > Preferences** menu item is disabled.

### To reproduce

1. Start Arduino IDE.
1. Watch the Arduino IDE window. As soon as the menu bar appears, select **File > Preferences**.
1. The "**Preferences**" dialog may be empty. Wait for it to populate.
1. Click the "**CANCEL**" button.
1. Open the **File** menu.

🐛 The **File > Preferences** menu item is disabled:

[image]

### Expected behavior

**File > Preferences** menu item is only disabled if there is a specific reason the preferences dialog should not be accessible to the user (The only reason I can think of is if the dialog is already open, but that should be handled by making the dialog modal).

### Arduino IDE version

5695fd8

### Operating system

Windows

### Operating system version

10

### Additional context

I bisected the introduction of the bug to https://github.com/arduino/arduino-ide/commit/7d6a2d5e3340aabc1e635d508509dc92a4000fdc (does not occur when using the build for https://github.com/arduino/arduino-ide/commit/6984c52b927260e13e23ae6eeb6f0b7488f7e12d).

---

Similar to https://github.com/arduino/arduino-ide/issues/1722 (note they were bisected to the same commit).

---

The fault does not occur if the IDE startup process is allowed to finish completely (as indicated by the disappearance of the busy spinner next to the mouse pointer).

---

The menu item is enabled in new Arduino IDE windows that are opened when the menu item is in the disabled state in the parent window.

---

Even when the menu item is in the disabled state, I can still open the "**Preferences**" dialog using the keyboard shortcut (<kbd>**Ctrl**</kbd>+<kbd>**,**</kbd>), or "**Arduino: Open Preferences**" command palette command.

---

I have experienced a similar fault with **Tools > WiFi101 / WiFiNINA Firmware Updater**, but have not found a way to reproduce it.

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

closed 2022-12-07 by kittaakos

sunzhongmeng, 2022-12-26:
I have the same problem.there is nothing in the Preferences dialog.
the OS is windows 10 Home (21H2)
[image]

per1234, 2022-12-28:
Hi @sunzhongmeng. That is a completely different fault. This issue is about the **File > Preferences** menu item being disabled. Your problem is the "**Preferences**" dialog being empty.

I saw your report about this on the Arduino Forum and am intending to investigate it:

https://forum.arduino.cc/t/arduino-ide-2-0-3-is-now-available/1061763/12

We can start over there and then create a dedicated bug report here if appropriate once we know more.

PR #1744 (2022-12-07): fix: `Preferences` menu enablement defect
### Motivation

To fix the `Preferences` menu enablement issue.

### Change description

 - Manually update the menus when executing the command.
 - Also disabled the menu when the settings dialog is opened.

### Other information

Closes #1735

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title and description are properly filled.
* [ ] Docs have been added / updated (for bug fixes / featur [... cut]
commit ed8ed151 (2022-12-07): fix: `Preferences` menu enablement defect

Manually update the menus when executing the command.

Also disabled the menu when the settings dialog is opened.

closes #1735