arduino/arduino-ide #1736: Serial Monitor scroll is incomplete when autoscroll is enabled

per1234, 2022-12-04:
### Describe the problem

The Arduino IDE "**Serial Monitor**" view has a "**Toggle Autoscroll**" control. When this is enabled, the view automatically scrolls down as the number of lines of data exceed the visible height of the view.

🐛 The autoscroll is not done completely, resulting in the most recently printed line only being partly visible.

### To reproduce

#### Equipment

- Any Arduino board capable of serial output.

#### Steps

1. Create a sketch that produces sufficient serial output to exceed the height of the "**Serial Monitor**" view:
   ```cpp
   void setup() {
     Serial.begin(9600);
     delay(10000);
     for (byte counter = 0; counter <= 100; counter++) {
       Serial.print("hello #");
       Serial.println(counter);
     }
   }
   void loop() {}
   ```
1. Connect the Arduino board to your computer.
1. Select the board and port in Arduino IDE.
1. Open the "**Serial Monitor**" view if it is not already open.
1. Select "**9600 baud**" from the dropdown baud rate menu at the top right corner of the "**Serial Monitor**" view.
1. If it is not already enabled, click the "**Toggle Autoscroll**" icon near the top left corner of the "**Serial Monitor**" view to enable autoscroll.
1. Immediately select the "**Serial Monitor**" tab in the bottom panel.
   **ⓘ** The "immediately" instruction, as well as the delay in the sketch, are required to avoid the demo's result being affected by a separate autoscroll-related bug `https://github.com/arduino/arduino-ide/issues/1724`
1. Wait until the sketch program's 10 s delay has passed.
   🐛 The line shown at the bottom of the "**Serial Monitor**" view is not `hello #100` as expected:
   [image]
1. Scroll the output field of the "**Serial Monitor**" view downward.
   🐛 The field was not scrolled all the way to the end of the output even though autoscroll was enabled.

### Expected behavior

The most recently printed line is always shown above the bottom edge of the Serial Monitor view when autoscroll is enabled.

### Arduino IDE version

5695fd8

### Operating system

Windows

### Operating system version

10

### Additional context

I bisected the issue to https://github.com/arduino/arduino-ide/commit/ac9cce1 / https://github.com/arduino/arduino-ide/pull/1662 (it does not occur when using the build from https://github.com/arduino/arduino-ide/commit/c0af1e6)

---

This bug also occurred in previous versions of Arduino IDE: https://github.com/arduino/arduino-ide/issues/972, but was fixed by https://github.com/arduino/arduino-ide/pull/1446

---

Additional reports:

(from time of the previous incarnation of the bug: https://github.com/arduino/arduino-ide/issues/972)

- https://github.com/arduino/arduino-ide/issues/1267
- https://github.com/arduino/arduino-ide/issues/1198
- https://github.com/arduino/arduino-ide/issues/812#issuecomment-1196261344
- https://forum.arduino.cc/t/ide-2-0-rc6-serial-monitor-window-clipping-on-bottom-line/984677
- https://forum.arduino.cc/t/serial-monitor-never-shows-the-last-line/1003734

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

sfe-SparkFro, 2022-12-20:
Adding my own observations in case it's helpful.

Testing with `2.0.4-nightly-20221220`, this issue persists. It appears to be that the latest chunk of data is not auto-scrolled, rather than just the last line. Here's some minimal code that demonstrates this behavior:

```
void setup() 
{
  Serial.begin(115200);
  
  Serial.println("Line 0");
 
  delay(100);
 
  Serial.println("Line 1");
  Serial.println("Line 2");
  Serial.println("Line 3");
  Serial.println("Line 4");
  Serial.println("Line 5");
}

void loop() {}
```

Testing on an ESP32, after pressing the reset button a couple times, this is always what Serial Monitor shows with auto scroll enabled:

[image]

Manually scrolling down shows the remaining lines as expected:

[image]

KurtE, 2022-12-21:
I am running daily build 1219, did not update today as did not see anything pulled in...

I am also finding my last things added are not being scrolled in to view.  In the case I was seeing I maybe added two lines at a time, and nothing showed up, until I used thumb wheel on mouse to scroll it.

Windows 11, Teensy MicroMod  Tensy board beta 0.58.3

fhb, 2023-01-07:
This bug made me believe there was something wrong with my dual rotary encoder or the Encoder library, spend half a day trying to fix the issue until I realised the buggy autoscroll is at fault! Well I learned a lot during this bug hunt. 

Mac Os Big Sur 11.7.2 (20G1020), Arduino IDE Version: 2.0.3, Date: 2022-12-05T09:29:06.685Z (4 weeks ago)

MK-Guy-I, 2023-01-19:
I face after have installed 2.0.3 (CLI Version 029.0 [76251df9]).

It was helpful to debug my state machine which drives my Lego train.

It's missing and does't help, now.

DanEfran, 2023-01-22:
I also wasted some time debugging a nonexistent problem due to this autoscroll bug. Further, it makes the kind of continuous control monitoring I was expecting to do almost impossible. This is a pretty serious problem in my opinion, and I hope it gets fixed soon.

logowe, 2023-01-24:
After getting used to scroll down all the time here I catched myself doing this in every other App - which lead to sometimes 'special' suprising results there ...
However suppose it helps to give away the colleague users of the Arduino IDE just by observing them for this strange behavour 😉

davidcliddell, 2023-02-01:
This really needs to be fixed. It is something everyone uses and the behaviour is difficult to deal with and is different from IDE 1. A CRITICAL bug

JGoor, 2023-02-15:
Please anyone. This is trivial. Just fix it. This should not be the type of bug that takes this much time to look into. I am considering an alternative IDE which is a pity, to say the least...

DavidSG2020, 2023-02-15:
Yes, it is annoying, and you have to wonder why it hasn't been fixed. Maybe it's a bug buried deep in a 3rd party component the developers are using?

sfe-SparkFro, 2023-02-16:
I've implemented a solution that seems to work and submitted a pull request #1889. For those who can, please give it a try and leave your feedback in the conversation of the pull request rather than here. No need to build from source, [beta testing instructions are here](https://github.com/arduino/arduino-ide/blob/main/docs/contributor-guide/beta-testing.md#testing-pull-requests). Thanks in advance!

KurtE, 2023-02-17:
FWIW: I downloaded the windows zip file Artifact from the checks tab of the PR and so far, it appears to work.

gohai, 2023-02-28:
This is a critical issue and needs to be addressed. I spent some time today attempting to chase down what looked to be a debouncing-issue on an digital input, when in reality I often was not looking at the most recent line of Serial data received.

I am hoping Arduino could release a fixed version promptly, since a working Serial monitor is so fundamental to this tool.

closed 2023-03-07 by kittaakos

ryaske, 2023-03-10:
I believe that this jumping scrolling behavior is somehow related to the existing bug: https://youtu.be/rAOLCMFk42k

per1234, 2023-03-11:
Hi @ryaske. Thanks for your report. Please try it with the latest nightly build of Arduino IDE. If the jumping still occurs when using that version of the IDE then please open a dedicated issue we can use to track the bug.

The nightly build download links are listed here:

https://www.arduino.cc/en/software#nightly-builds

sfe-SparkFro, 2023-03-13:
I can confirm `2.0.4-nightly-20230312` has resolved the incomplete autoscroll issue. The jumping scrolling issue looks to be caused by the horizontal scroll bar popping in and out for whatever reason; I've not seen it myself, but that is a different issue.

PR #1928 (2023-03-07): fix: scroll to the bottom after the state update
### Motivation

An alternative version of #1889. This PR also fixes #1724.

Scroll to the bottom fix:

https://user-images.githubusercontent.com/1405703/222668221-46b68d3b-ce4b-48ea-ba67-35f6d22b8667.mp4

Update widget content and scroll position after the widget shows:

https://user-images.githubusercontent.com/1405703/222668321-274bd881-f707-498f-bf3b-7b77af29b7a4.mp4

### Change description

Scroll to the bottom of the monitor widget after the state update.

### Other information

Closes #1724
Closes #1736

### Reviewer checklist

* [ ] PR ad [... cut]
commit fa9777e5 (2023-03-07): fix: scroll to the bottom after the state update

Closes #1736