arduino/arduino-ide #1737: Add support for complex sketch hierarchy

MathieuAndrade, 2022-12-04:
### Describe the request

Add support for complex sketch hierarchy when sketch has multiple C CPP and H subfiles.

Folder/
├── Sketch Folder/
│   ├── Sketch.ino
│   └── src/
│       ├── sketch-utils.h
│       ├── sketch-utils.cpp
│       └── [...]
└── README

### Describe the current behavior

Actually, the IDE does not open files in sub folder even if file is present and is included in main sketch and we cannot open manually files in tabs. When we create a new tab the IDE create systematically a new file.

In Sketchbook explorer the IDE show files in subfolder only if sketch is in Sketchbook location  ->

[image]

If the opened sketch isn't in Sketchbook location the Sketchbook explorer does not show files -> 

[image]

(Related to https://github.com/arduino/arduino-ide/issues/1461#issuecomment-1250886642)

So on this situation we cannot open subfiles of sketch 😕 

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

weswitt, 2022-12-05:
+1000 for this. but also allow for "..\..\" paths out to shared cpp & h files. just like what make supports

per1234, 2022-12-05:
Hi @weswitt. Thanks for your suggestion. Not automatically opening sketch files from the `src` subfolder is the intentional design of Arduino IDE. This allows sketch authors to create complex sketches, while only immediately exposing the user to the files they are most likely to need access to.

A good example of this is the popular **Marlin** 3D printer firmware. The average user of that sketch only needs access to the files that contain the configuration settings, so those are placed in the root of the sketch folder:

https://github.com/MarlinFirmware/Marlin/tree/bugfix-2.1.x/Marlin

The hundreds of other files of the sketch were intentionally placed under the `src` subfolder because the average user does not need access to them.

As you mentioned, Arduino IDE 2.x does allow the files from the `src` subfolder of the sketch to be opened via the **Sketchbook** view if the "**Show files inside Sketches**" preference is enabled. So if you want this capability, simply store your sketches in the sketchbook folder.

MathieuAndrade, 2022-12-05:
Hello @per1234 

> The hundreds of other files of the sketch were intentionally placed under the `src` subfolder because the average user does not need access to them.

Ok this can make sense and i agree.

> As you mentioned, Arduino IDE 2.x does allow the files from the `src` subfolder of the sketch to be opened via the **Sketchbook** view if the "**Show files inside Sketches**" preference is enabled. So if you want this capability, simply store your sketches in the sketchbook folder.

Why intentionally make this?
In some of environnement devs cannot have **all** sketchs in **only** one folder, this IDE is a very good upgrade of the first version, but with this limitation he becomes useless in some of the use cases. Mostly when we cannot open this subfiles manually in tabs. 
Theia IDE has a view that shows the files of the current scope, maybe can you provide a checkbox preference to (re)allow this view ?