arduino/arduino-ide #1740: "Updatable" filter not set in Boards/Library Manager during "INSTALL MANUALLY" operation if other view open in side panel

per1234, 2022-12-05:
### Describe the problem

Arduino IDE automatically checks for available updates to boards platforms and libraries installed via **Boards Manager** and **Library Manager**. If an update is available, a notification dialog appears with the following options:

- **LATER**
- **INSTALL MANUALLY**
- **INSTALL ALL**

"**INSTALL MANUALLY**" opens the **Boards Manager** or **Library Manager** view with the "**Type**" filter set to "**Updatable**". This allows the user to evaluate the available updates and install them selectively.

If another view was already open in the Side Panel, the "**Type**" filter is left set to the default "**All**" when the **Boards Manager** or **Library Manager** view is opened.

🐛 "**INSTALL MANUALLY**" will not provide the user with a convenient path to update dependencies.

### To reproduce

#### Set up

An outdated version of a library is installed in order to trigger the update offer dialog:

1. Open the **Library Manager** view if it is not already open.
1. In the "**Filter your search**" field, type `arduinocloudthing`
   **ⓘ** This is just an arbitrarily selected library with no special significance.
1. Select "**1.7.2**" from the version menu on the "**ArduinoCloudThing**" entry.
1. Click the "**INSTALL**" button.
1. An "**Install library dependencies**" dialog will now appear. Click the "**INSTALL WITHOUT DEPENDENCIES**" button.
1. Wait for the "**ArduinoCloudThing**" library to be installed successfully.

#### Demo

1. Select **File > New Sketch** from the Arduino IDE menus.
   **ⓘ** This is done to ensure the **Library Manager** view has the default "**Type**" filter setting.
1. Click the folder icon on the Activity Bar to open the "**Sketchbook**" view in the Side Panel.
1. Press the <kbd>**Ctrl**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> keyboard shortcut (<kbd>**Command**</kbd>+<kbd>**Shift**</kbd>+<kbd>**P**</kbd> for macOS users) to open the "**Command Palette**".
1. Select the "**Arduino: Check for Arduino Updates**" command from the menu.
1. Wait for the "**Updates are available for some of your libraries.**" notification dialog to appear.
1. Click the "**INSTALL MANUALLY**" button.

🐛 The **Library Manager** view opens with the "**Type**" filter set to "**All**":

[image]

### Expected behavior

The "**Type**" filter is always 

### Arduino IDE version

3.0.3

### Operating system

Windows

### Operating system version

10

### Additional context

I bisected the introduction of the bug to ac9cce1 / https://github.com/arduino/arduino-ide/pull/1662 (does not occur when using the build for the previous commit 3ad6609 or older).

---

The "**Arduino: Check for Arduino Updates**" command palette command was used in the demo as a convenient way to trigger the update notification. The bug also affects the more commonly used update notification on IDE startup.

---

I used the technique of manually opening a view in the side panel in the demo in order to make it as simple as possible. However, the fault also occurs in the more likely use cases when the panel was open for other reasons, such as:

- The panel open state was restored for the workspace on startup due to it having been open during the previous IDE session.
- Both Boards and Library Manager updates were available and the panel was previously opened for a manual update operation via the other view ([real life example](https://forum.arduino.cc/t/arduino-2-0-3-update-boards/1061726/1).

---

The fault does not occur when opening the view with filter configured via [a `librarymanager`/`boardsmanager` link](https://arduino.github.io/arduino-cli/dev/sketch-specification/#libraryboards-manager-links).

---

Originally reported by @KurtE at https://forum.arduino.cc/t/arduino-2-0-3-update-boards/1061726

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

kittaakos, 2022-12-07:
It's working as expected on macOS with 2.0.3. I give it a try on Windows.

https://user-images.githubusercontent.com/1405703/206179476-f9ab1b65-4984-4b52-b586-b2849d21e5f6.mp4

kittaakos, 2022-12-07:
It's also working on my Windows env with 2.0.3.

https://user-images.githubusercontent.com/1405703/206186735-d81d7916-39e2-4282-9022-c6832c09d7bc.mp4

kittaakos, 2022-12-07:
Are there any exceptions in the DevTools console?

per1234, 2022-12-07:
Hi @kittaakos. You forgot to do step 2 in the instructions:

>2. Click the folder icon on the Activity Bar to open the "Sketchbook" view in the Side Panel.

A required condition is that another view is previously open in the Side Panel.

KurtE, 2022-12-07:
Note: I again just reproduced it using last nights build:

In this case, I started up the IDE.  It came up with saying I had both boards and libraries needing updates.  
It was also showing that it was downloading the teensy boards... (which takes a long time).

I clicked on the update libraries, which did not show anything until the Teensy boards file finished downloading. Which felt like a few minutes.  At that point I had the usual SDFat ...

So then I clicked manual on boards and ended up with window looking like:
[image]

kittaakos, 2022-12-07:
Thank you for the help 🙏 

> 1. Select **File > New Sketch** from the Arduino IDE menus.
>    **ⓘ** This is done to ensure the **Library Manager** view has the default "**Type**" filter setting.

It turned out that the key point was the new sketch and that the _lib manager_  view was not yet opened in IDE2. There was a race condition.

closed 2022-12-08 by kittaakos

KurtE, 2022-12-08:
I assume it did not make it into today's daily build?
[image]

Will try it again tomorrow.

kittaakos, 2022-12-09:
> it did not make it into today's daily build?

Yes, it did, but you're using `20221208` and the [latest nightly is `2.0.4-nightly-20221209`](https://github.com/arduino/arduino-ide/actions/runs/3654083918/jobs/6174192969#step:7:270).

KurtE, 2022-12-09:
Thanks, 

Note: we are probably in completely different time zones, and when I posted that last response, it was on the morning of the 8th.
Now it is early morning of the 9th and I have ow downloaded the new one and it works!

Thanks

PR #1748 (2022-12-08): fix: list view filtering when opening the view
### Motivation

Fix the search options behavior of the library and boards manager widget when programmatically opening them.

### Change description

 - When the widget is opened the first time with any search options, the widget might miss the refresh event as it does not happen at instantiation time. This PR ensures that all list widget refresh events wait until the React component is rendered and is visible in the UI.
 - This change also ensures that the debounced search will run with the most recent search options by setting the `trailing` pro [... cut]
commit f8c01e37 (2022-12-08): fix: list view filtering when opening the view

When the widget is opened the first time with any search options,
the widget might miss the refresh event as it does not happen at
instantiation time. This PR ensures that all list widget refresh
events wait until the React component is rendered and is visible
in the UI.

This change also ensures that the debounced search will run with
the most recent search options by setting the `trailing` property
of the debounced function to `true`.

Closes #1740