arduino/arduino-ide #1741: Search on All Tabs

muchahagrande, 2022-12-05:
### Describe the request

Hi, I'd like to be able to use the search/replace on all tabs as it is possible on Arduino IDE 1.x series.

### Describe the current behavior

I can choose to search on selection, but I can't search on multiple tabs at once. So I have to search on each tab separately.

### Arduino IDE version

2.0.3

### Operating system

Linux

### Operating system version

Ubintu 20.04

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-05:
Hi @muchahagrande. Thanks for your suggestion.

I'm happy to tell you that this is already possible. Simply click the magnifying glass icon on the "Activity Bar" on the left side of the Arduino IDE window to open the "**Search**" panel. I think you'll find it offers a very capable search and replace functionality.

[image]

closed 2022-12-05 by per1234

muchahagrande, 2022-12-06:
I didn't see it until you mentioned it :)
This is what I was looking for... thank you for your reply.
Cheers...