arduino/arduino-ide #1751: Changing board does not update partition scheme selection - IDE 2.0.4

illigtr, 2022-12-08:
### Describe the problem

Changing board does not update partition scheme selection in the GUI - IDE 2.0.4

### To reproduce

Example using ESP32 board manager installed.

Using Menu/Tools/Board/esp32/ select any board
Then view the partition schemes available for that board under Menu/Tools/Partition Scheme.
Note the list.

Now select a different board 
The partition schemes do not update correctly from the boards.txt file.

Used process monitor to ensure the correct boards.txt is being read when the IDE first starts.

### Expected behavior

The GUI should update and present the specific list of partition schemes for the selected board. It does not.

One can also test this by editting the boards.txt and adding some custom schemes. They will not appear in the GUI. However, these feature works perfectly in IDE 1.8.19

### Arduino IDE version

2.0.4-nightly-20221208

### Operating system

Windows

### Operating system version

11

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-10:
Hi . Thanks for taking the time to submit an issue.

Even though your report implies that the **Tools > Partition Scheme** menu never reflects differences in the menu options from one board to another, my experiment showed this is not so:

1. Select **Tools > Board > esp32 > ESP32 Dev Module** from the Arduino IDE menus.
1. Open the **Tools > Partition Scheme** menu. Note there is a "**RainMaker**" menu item, [as expected](https://github.com/espressif/arduino-esp32/blob/2.0.5/boards.txt#L646).
1. Select **Tools > Board > esp32 > ESP32 Wrover Module** from the Arduino IDE menus.
1. Open the **Tools > Partition Scheme** menu. Note there is no "**RainMaker**" menu item, [as expected](https://github.com/espressif/arduino-esp32/blob/2.0.5/boards.txt#L935-L962).

So in fact the menu does reflect the correct options from the unmodified board definition.

The problem becomes more clear from your statement here:

>One can also test this by editting the boards.txt and adding some custom schemes. They will not appear in the GUI. 

I see we have another report about this at https://github.com/arduino/arduino-ide/issues/1030.

It is best to have only a single issue per subject so we can consolidate all relevant discussion to one place, so I'll go ahead and close this in favor of the other.

If you end up with additional information to share, feel free to comment in the other thread.

closed 2022-12-10 by per1234

illigtr, 2022-12-10:
Thank you for taking the time to respond so quickly. My search for similar
issues did not find anything and I should have used the “boards.txt”
keyword.

I have tried the workaround given in #1030 (delete
\Appdata\Roamin\arduino-ide) and it has resolved my issue with custom
entries made in boards.txt

*From:* per1234 ***@***.***
*Sent:* December 10, 2022 06:09
*To:* arduino/arduino-ide ***@***.***>
*Cc:* illigtr ***@***.***>; Author ***@***.***>
*Subject:* Re: [arduino/arduino-ide] Changing board does not update
partition scheme selection - IDE 2.0.4 (Issue #1751)

Hi . Thanks for taking the time to submit an issue.

Even though your report implies that the *Tools > Partition Scheme* menu
never reflects differences in the menu options from one board to another,
my experiment showed this is not so:

   1. Select *Tools > Board > esp32 > ESP32 Dev Module* from the Arduino
   IDE menus.
   2. Open the *Tools > Partition Scheme* menu. Note there is a "*RainMaker*"
   menu item, as expected
   <https://github.com/espressif/arduino-esp32/blob/2.0.5/boards.txt#L646>.
   3. Select *Tools > Board > esp32 > ESP32 Wrover Module* from the Arduino
   IDE menus.
   4. Open the *Tools > Partition Scheme* menu. Note there is no "
   *RainMaker*" menu item, as expected
   <https://github.com/espressif/arduino-esp32/blob/2.0.5/boards.txt#L935-L962>
   .

So in fact the menu does reflect the correct options from the unmodified
board definition.

The problem becomes more clear from your statement here:

One can also test this by editting the boards.txt and adding some custom
schemes. They will not appear in the GUI.

I see we have another report about this at #1030
<https://github.com/arduino/arduino-ide/issues/1030>.

It is best to have only a single issue per subject so we can consolidate
all relevant discussion to one place, so I'll go ahead and close this in
favor of the other.

If you end up with additional information to share, feel free to comment in
the other thread.

Adrianotiger, 2022-12-15:
Boards info are cached in:
`%APPDATA%\arduino-ide\Local Storage\leveldb`

If you delete the *.ldb files inside this folder it works (at least for the partition schemes). But you need to set again all settings for your board.