arduino/arduino-ide #1754: COM port is blocked

RudolfAtHome, 2022-12-08:
### Describe the problem

Since version 2.0.3: The COM port is blocked even if the "Serial Monitor" is not used!
I use an external program as a replacement for the Serial Monitor. That doesn't work anymore. I classify that as a mistake!

Up to and including V 2.0.2 the COM port was not blocked and I could use my external program.
Now I have to close the IDE - that doesn't work at all.

### To reproduce

- IDE is open and I can download a sketch,
- After that my external program has no access to the related COM port!
- I must close the IDE to deal with my program

### Expected behavior

Same as in V 2.0.2 and bevor: The IDE has to release the COM port after download.
If one uses the integrated Serial Monitor, only than the port must be used too. 

### Arduino IDE version

Arduino IDE 2.0.3

### Operating system

Windows

### Operating system version

Windows 10 personal

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-10:
Hi @RudolfAtHome. Thanks for your report. I am not able to reproduce this. Please provide more information:

- Please provide detailed step by step instructions I can follow to reproduce the problem.
- Which Arduino board?(s) are you using when you encounter this problem?
- Does the problem occur consistently, or only intermittently?
- Do you ever open the Arduino IDE Serial Monitor or Serial Plotter during the session (regardless of whether they are closed at the time you try to open the port using the external program)?

PaulStoffregen, 2022-12-10:
@RudolfAtHome - Does killing the serial-discovery.exe process in Task Manager (select it and click "End task") give you back access to your COM port?

[image]

RudolfAtHome, 2022-12-10:
**Sorry - it's not an error!**
  - If the serial monitor is open in the IDE after a download, the COM port is blocked. The serial monitor does not have to be activated.
  - When I close the Serial Monitor, the COM port is available again for all other programs.
I can live with that.

closed 2022-12-10 by per1234