arduino/arduino-ide #1755: Changing font in Arduino IDE 2.X

LazaroFilm, 2022-12-08:
### Describe the request

I would like to change the font in Arduino IDE to a font with ligatures such as JetBrains Mono, or Fira code. The ligatures make multi-character simbols such as `=>` or `==` easier to read. I can't find any option to select a new font. 

### Describe the current behavior

You can only change the theme or the font size. 

### Arduino IDE version

2.0.3

### Operating system

macOS

### Operating system version

13.0.1

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-09:
Hi @LazaroFilm. The font can be changed in the Arduino IDE advanced settings:

https://github.com/arduino/arduino-ide/blob/main/docs/advanced-usage.md#advanced-settings

If you have any questions or problems while changing the font, please post over on the Arduino Forum. I'm sure we'll be able to help you out over there:

https://forum.arduino.cc/c/software/arduino-ide-2-0/93

closed 2022-12-09 by per1234

Xylopyrographer, 2024-11-16:
Found that in Arduino IDE v 2.3.3 this can be done.

Open the Advanced settings:

Search for "open settings"
Select "Preferences:Open Settings (UI)"
Search settings for "ligatures"

Find: 
Editor: Font Ligatures
Configures font ligatures or font features. Can be either a boolean to enable/disable ligatures or a string for the value of the CSS 'font-feature-settings' property.

Select "Edit in settings.json"

In settings.json", find the line:
    `"editor.fontLigatures": false,`
and change it to
    `"editor.fontLigatures": true,`

This assumes you have set as your primary Editor font one that contains "programmers ligatures" such as JetBrains Mono or Fira Code.

If such a font is installed, change the Editor font list to use it. To do so, in the same `settings.json` file, search for `editor.fontFamily:` and edit it so the font you want is listed first. It should look something like:

`"editor.fontFamily": "JetBrains Mono, Menlo, Monaco, 'Courier New', monospace",`

Going further, you can also specify the font used by the integrated Terminal by searching in the `settings.json` file for `"terminal.integrated.fontFamily"` and specifying it first in the list there. For example:

`"terminal.integrated.fontFamily": "JetBrains Mono, Menlo, Monaco, 'Courier New', monospace",`

Note that ligatures cannot be used in the Terminal.