arduino/arduino-ide #1765: The 2.0.x version of intellisense is not working well.

blaticslm, 2022-12-12:
### Describe the request

When I hold CTRL, I can immediately click the library to see the source code without any problem like vscode. If the intellisense is indexing the libraries and then I can CTRL+ left click, please have a constant progress bar at the lower left corner to indicate it is in progress because this is not even working well! 

### Describe the current behavior

This is so annoying that sometimes it works and sometimes are not!

When I held CTRL to the library header file, I can't click to see the details! BUT SOMETIMES it works again. I have no idea how is this functionality frustrating me so much. 
[image]
[image]

### Arduino IDE version

2.0.2

### Operating system

Windows

### Operating system version

10

### Issue checklist

- [X] I searched for previous requests in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the feature was still missing when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My request contains all necessary details

per1234, 2022-12-13:
Hi @blaticslm. Please try this:

1. Get Arduino IDE into the "can't click to see the details" state.
1. Select **Sketch > Verify/Compile** from the Arduino IDE menus.
1. Wait for the compilation to finish.
1. Try the <kbd>**Ctrl**</kbd>+**left click** operation again. 

Please comment here to let me know whether the "can't click to see the details" problem still occurs after you compiled the sketch.

blaticslm, 2022-12-16:
OK...I will try to do it... it was so frustrated.

per1234, 2023-01-30:
Hi @blaticslm. We are still waiting for you to provide the information requested in my previous reply so that we can proceed with the investigation or evaluation of this.

Did you ever get a chance to try that? If so, what were the results?

blaticslm, 2023-01-30:
> Hi @blaticslm. We are still waiting for you to provide the information requested in my previous reply so that we can proceed with the investigation or evaluation of this.
> 
> Did you ever get a chance to try that? If so, what were the results?

I did try, and now I can fully utilize IntelliSense. But I just feel really bad that why this IDE needs to be verified/compiled first to activate IntelliSense. In PlatformIO in vscode, I can just import whatever library first, and directly go to see their source code API.

closed 2023-03-12 by per1234