arduino/arduino-ide #1769: Doesn't work with SOCKS proxy

wh201906, 2022-12-13:
### Describe the problem

I have a SOCKS proxy on port `1233` and a http proxy on port `1234`. No matter which proxy type I select in Arduino IDE(`SOCKS` or `HTTP`), if I set the port to `1234`, everything works fine, if I set the port to `1233`, the network error occurs. **It seems like the Arduino always uses the http proxy regardless of the proxy type I selected.**
Plus, If I configure a http proxy in settings, it will be shown as SOCKS proxy the next time I open the Arduino IDE. But the `arduino-cli.yaml` still shows that it's a http proxy.

### To reproduce

1. Set a SOCKS proxy on port `n` and a http proxy on port `n+1`
2. In Arduino, set the SOCKS proxy on port `n`, the network error occurs
3. Set the http proxy on port `n+1`, the network works fine
4. Set the SOCKS proxy on port `n+1`, the network works fine (which is unexpected)
5. Set the http proxy on port `n`, the network error occurs

### Expected behavior

The SOCKS proxy should work.

### Arduino IDE version

Version: 2.0.3 Date: 2022-12-05T09:30:25.331Z CLI Version: 0.29.0 [76251df9]  Copyright © 2022 Arduino SA

### Operating system

Windows

### Operating system version

Windows 10

### Additional context

I have checked the proxy and I'm sure port `1233` only accepts SOCKS and `1234` only accepts http

### Issue checklist

- [X] I searched for previous reports in [the issue tracker](https://github.com/arduino/arduino-ide/issues?q=)
- [X] I verified the problem still occurs when using the latest [nightly build](https://www.arduino.cc/en/software#nightly-builds)
- [X] My report contains all necessary details

per1234, 2022-12-14:
Hi @wh201906. Thanks for your report. I was not able to reproduce it.

The instructions you provided are too vague for me to be sure I am correctly reproducing what you have done. Please provide a set of very detailed and clear instructions that I can follow for a minimal reproduction of the bug and I will continue with the investigation.

wh201906, 2022-12-14:
Thanks for your reply. Here are the steps:
<details>
<summary>steps</summary>

1. Set a socks proxy on 127.0.0.1:1233 and a http proxy on 127.0.0.1:1234
2. Open Arduino IDE, go to preference->network
3. Set the proxy like this
[image]
4. Click OK, then the net work error occurs
[image]
5. Set the proxy like this
[image]
6. Click OK, the IDE shows the downloading process
[image]
7. Set the proxy like this
[image]
8. Click OK, the IDE shows the downloading process
[image]
**This is weird because port 1234 only accepts http proxy**
9. Set the proxy like this
[image]
10. Click OK, then the net work error occurs
[image]

</details>

# Additional information
When changing the proxy in the Arduino IDE, I can see the configuration is applied to the `arduino-cli.yaml`
[image][image][image]
I guess it's something wrong in the Arduino-cli?
Plus, when dumping the config in arduino-cli, I can't see the network config.
[image]
Please tell me if you need more information.

per1234, 2022-12-16:
Thanks so much for that additional information @wh201906. That was very helpful. I was able to reproduce the bugs you reported here and track down the cause of the primary bug:

## SOCKS Proxy Doesn't Work

> 4. Click OK, then the net work error occurs

> 8. Click OK, the IDE shows the downloading process
>[...]
>This is weird because port 1234 only accepts http proxy

> My proxy software shows the Arduino IDE is still using http proxy even if I just set a socks proxy

> I guess it's something wrong in the Arduino-cli?

### Explanation

It is actually a bug in Arduino IDE that is causing all these symptoms. The problem is it uses the `socks` scheme in the URL when it sets the Arduino CLI proxy configuration via the `network.proxy` key of `arduino-cli.yaml`. The `net/http` Go package used by Arduino CLI requires the scheme to instead be `socks5`:

https://pkg.go.dev/net/http#Transport

I was not able to track down the code responsible for the behavior, but it seems that HTTP is used as a fallback when the scheme in the proxy URL is not recognized. This is why you found the HTTP proxy works even when you set the radio button to "**SOCKS**" in the Arduino IDE preferences and it sets the `network.proxy` key to a `socks://` URL.

### Workaround

1. Open the following file in a text editor (I realize you already know where it is, but I'll provide comprehensive instructions for the benefit of any other users who might be affected):
   - If you are using **Windows**:
     ```text
     C:\Users\<username>\.arduinoIDE\
     ```
   - If you are using **Linux**:
     ```text
     ~/.arduinoIDE/
     ```
   - If you are using **macOS**:
     ```text
     ~/.arduinoIDE/
     ```
1. Change the `socks://` part of the URL in the `network.proxy` key to `socks5://`
   For example, if the file contained this:
   ```yaml
   network:
     proxy: socks://127.0.0.1:1233
   ```
   It should be changed to this:
   ```yaml
   network:
     proxy: socks5://127.0.0.1:1233
   ```
1. Save the file.
1. If Arduino IDE is running, select **File > Quit** from the menus to close all windows.
1. Start Arduino IDE.

## Proxy Configuration Set by Arduino IDE not Shown in `arduino-cli config dump` Output

>After setting the proxy in Arduino IDE, I can't see the network config when dumping the config in arduino-cli.

Arduino CLI recognizes `arduino-cli.yaml` configuration files in several locations:

https://arduino.github.io/arduino-cli/latest/configuration/#locations

The dedicated `~/.arduinoIDE/arduino-cli.yaml` file used by Arduino IDE is not in one of these locations (unless you happen to be running the `arduino-cli` command from that folder). So if you run `arduino-cli config dump` you will get the effective configuration of Arduino CLI (either the defaults, or taken from an `arduino-cli.yaml` file if it exists in one of the recognized locations.

You can set the location of the file via the [--config-file](https://arduino.github.io/arduino-cli/latest/commands/arduino-cli/#options) command line flag, so if you run this command you would see the Arduino IDE configuration in the output:

```text
arduino-cli config dump --config-file $Env:HOMEDRIVE$Env:HOMEPATH/.arduinoIDE/arduino-cli.yaml
```

## "Preferences" Dialog Shows Incorrect Proxy Type

> Open Arduino IDE, go to Preferences->Network, the proxy type switches from http to socks

I am able to reproduce this. I think it will be more effective and efficient to dedicate this issue exclusively to the incorrect `socks` URL scheme used when configuring Arduino CLI for a SOCKS proxy, so I have created a separate issue to track the bug with the incorrect radio button selection in the "**Preferences**" dialog: https://github.com/arduino/arduino-ide/issues/1775

wh201906, 2022-12-16:
Thanks for handling these bugs!

wh201906, 2022-12-16:
> The `net/http` Go package used by Arduino CLI requires the scheme to instead be `socks5`

@per1234 Hi. Can I make a PR to make a fix for it?

per1234, 2022-12-16:
Yes, a PR would be welcome!

PR #1776 (2022-12-21): Replace socks with socks5 in proxy protocol
### Motivation
Fixes #1769
The net/http package in arduino-cli supports `socks5` as scheme rather than `socks`. This PR replaces `socks` with `socks5` in proxy protocol to make socks proxy work.

### Change description
1. Replaces `socks` with `socks5` in proxy protocol
2. Show `SOCKS5` in `Preferences`->`Network`

### Other information

### Reviewer checklist

* [ ] PR addresses a single concern.
* [ ] The PR has no duplicates (please search among the [Pull Requests](https://github.com/arduino/arduino-ide/pulls) before creating one)
* [ ] PR title a [... cut]
closed 2022-12-21 by kittaakos